dotintent/react-native-ble-plx #948: getting null device name while calling startDeviceScan

scrapecoder, 2022-01-19:
const scanAndConnect = () => {
    bleManager.startDeviceScan(
      null,
      {allowDuplicates: false},
      (error, device) => {
        if (error) {
          if (error.message === 'BluetoothLE is powered off') {
            bleManager.enable();
          }
          // Handle error (scanning will be stopped automatically)
          return;
        }

        // Check if it is a device you are looking for based on advertisement data
        // or other criteria.

        if (device && device.name) {
          console.log('device name====>', device.name);
        }
        console.log('device-<', device);
      },
    );
    };

`"id":"40:1E:8E:D2:E5:7E",
    "isConnectable":null,
    "localName":null,
    "manufacturerData":"TAAQBUccC65x",
    "mtu":23,
    "name":null,
    "overflowServiceUUIDs":null,
    "rssi":-37,
    "serviceData":null,
    "serviceUUIDs":null,
    "solicitedServiceUUIDs":null,
    "txPowerLevel":null`

* Library version: '^2.3.0'
* Platform: Android.

mciechanowicz, 2022-01-28:
HI,
can you use some external scanning application to see if the name is there or it is also null?

mreda7, 2022-03-22:
Hi @scrapecoder , I have the same problem here . Did you find a solution for this ?

Robiullah2244, 2022-05-23:
Same issue

disconnect-ed, 2022-09-20:
Same issue

yosukapro, 2023-02-15:
+1 android

abhinavshm95, 2023-05-11:
+1 iOS

iSaBo96, 2023-05-27:
+1 iOS

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

dominik-czupryna-withintent, 2023-11-02:
We have added a scanning option for BT 5 advertisers (disable legacy scan - https://github.com/dotintent/react-native-ble-plx/wiki/Device-Bonding) and name mapping for Android.

closed 2023-11-02 by dominik-czupryna-withintent