dotintent/react-native-ble-plx #949: Feature Request

WaldmannIoTDevelopers, 2022-01-21:
## Prerequisites

- [X] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Current Behavior

Scanning for Advertisings:
On Android currently only scan response data is returned from startDeviceScan() - see settings below -

Example for our specific device sending out following data:
Manufacturer Data Struct within Default Advertising Data: 
**04:A4:00:8C:21:0E:B9**
Another Manufacturer Data Struct within Scan Response Data: A4:04:**01:CD:09:47:A8:0C:E9:6F:FF:FF:F7:FF:FF.FE:01:01:23:4D:14:DC:01:80:07:AC:90:3E:19**

But all (!) Scan Result Events are only including the data of the scan response, but not the data of the default advertising data packet.

On iOS we are getting a combined packet, including bot data parts, like this:
**04:A4:00:8C:21:0E:B9****01:CD:09:47:A8:0C:E9:6F:FF:FF:F7:FF:FF.FE:01:01:23:4D:14:DC:01:80:07:AC:90:3E:19**

This is a bit in a strange format (because it strips A4:04 when appending to first data packet), but i works for us.

Here is how i called
startDeviceScan(
  null,
  { callbackType: ScanCallbackType.AllMatches, scanMode: ScanMode.LowLatency, allowDuplicates: true },
  (error: BleError | null, scannedDevice: Device | null) => {
    if (error) {
      console.error(error);
      return;
    }
    if (!scannedDevice) return;
    console.log(scannedDevice);
  }
);

## New behavior

Could you please also implement to provide advertising data (not only scan response) for android?

For us it would also be sufficient in a first step, if you provide rawBytes data
byte[] | getBytes() Returns raw bytes of scan record. (https://developer.android.com/reference/android/bluetooth/le/ScanRecord)

### Breaking changes

Does a feature breaks compatibility with previous version?
-> No, if you provide differnent data (advertising + scan response) in multiple scanresult events within "manufacturerData" property.
-> Also no, if you provide another optional property in scan events (for example "rawData") for providing raw bytes

Really appreciate your work! Thanks in advance.

jfamousket, 2022-02-09:
Hey you found anyway to go around this, I also need advertising data and it's currenlty `null` on android. Also looks like support is slow around here

luhaimin, 2022-04-22:
I also encountered the same problem. Why not give priority to such an important problem?

commit beba31be (2023-11-14): feat: resolves #949 - add additional information with raw scan record
closed 2023-11-14 by dominik-czupryna-withintent