dotintent/react-native-ble-plx #952: Unable to Initialize Library in React Hook

brianbancroft, 2022-01-29:
## Expected Behaviour

I seek to run the "hello world" example, as illustrated in your quick start example [here](https://github.com/dotintent/react-native-ble-plx/blob/master/INTRO.md). My desired goal is to a point where I can console log a list of available devices. 

## Current Behaviour

At this time, I am unable to initialize the library.

When running `const foo = new BleManager();` in a React component file, I run into the following error:

`TypeError: null is not an object (evaluating '_BleModule.BleModule.createClient')`

### Steps to Reproduce

I am using Expo as an environment with Android Studio. The steps for reproduction are:

#### Step 1: Initialize an Expo RN app and add dependency

- In a terminal, type `npx expo init hello-bt-error`

It doesn't matter if it's JS or TS. The error that comes out is still consistent.

- Add the react-native-ble-plx dependency. 

#### Step 2: Initialize the lib

Outside of the App.js/App.tsx component, add the following lines:

```js
import { BleManager } from 'react-native-ble-plx';

export const manager = new BleManager();
```

That's sufficient to get the full error.

#### Step 3: Run the app either on a physical android device, or an emulated one.

In my case, I've emulated a pixel 5, and my personal hardware is a pixel 3. Neither work. 

### Context

* Node JS: 16.13.1 (lts/gallium)
* Library version: 2.0.3
* Development Platform: Macbook Pro. MacOS Big Sur 11.6
* Android Studio

<details>
  <summary>Android Studio Version Details (click to expand)</summary>

```
Android Studio Arctic Fox | 2020.3.1 Patch 4
Build #AI-203.7717.56.2031.7935034, built on November 20, 2021
Runtime version: 11.0.10+0-b96-7281165 x86_64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
macOS 11.6
GC: G1 Young Generation, G1 Old Generation
Memory: 1280M
Cores: 16
Registry: external.system.auto.import.disabled=true
Non-Bundled Plugins: org.jetbrains.kotlin
```

</details>

<details>
  <summary>JS Logs (click to expand)</summary>

```
[Unhandled promise rejection: TypeError: null is not an object (evaluating '_BleModule.BleModule.createClient')]
at node_modules/@react-navigation/native/src/useThenable.tsx:25:13 in resolve

BleManager  [Function BleManager]
BleManager  [Function BleManager]

TypeError: null is not an object (evaluating '_BleModule.BleModule.createClient')

This error is located at:
    in BluetoothTest (created by TabOneScreen)
    in RCTView (created by View)
    in View (created by View)
    in View (created by TabOneScreen)
    in TabOneScreen (created by SceneView)
    in StaticContainer
    in EnsureSingleNavigator (created by SceneView)
    in SceneView (created by BottomTabView)
    in RCTView (created by View)
    in View (created by Screen)
    in RCTView (created by View)
    in View (created by Background)
    in Background (created by Screen)
    in Screen (created by BottomTabView)
    in RNSScreen (created by AnimatedComponent)
    in AnimatedComponent
    in AnimatedComponentWrapper (created by Screen)
    in MaybeFreeze (created by Screen)
    in Screen (created by MaybeScreen)
    in MaybeScreen (created by BottomTabView)
    in RNSScreenContainer (created by ScreenContainer)
    in ScreenContainer (created by MaybeScreenContainer)
    in MaybeScreenContainer (created by BottomTabView)
    in RCTView (created by View)
    in View (created by SafeAreaInsetsContext)
    in SafeAreaProviderCompat (created by BottomTabView)
    in BottomTabView (created by BottomTabNavigator)
    in Unknown (created by BottomTabNavigator)
    in BottomTabNavigator (created by BottomTabNavigator)
    in BottomTabNavigator (created by SceneView)
    in StaticContainer
    in EnsureSingleNavigator (created by SceneView)
    in SceneView (created by SceneView)
    in RCTView (created by View)
    in View (created by DebugContainer)
    in DebugContainer (created by MaybeNestedStack)
    in MaybeNestedStack (created by SceneView)
    in RNSScreen (created by AnimatedComponent)
    in AnimatedComponent
    in AnimatedComponentWrapper (created by Screen)
    in MaybeFreeze (created by Screen)
    in Screen (created by SceneView)
    in SceneView (created by NativeStackViewInner)
    in RNSScreenStack (created by ScreenStack)
    in ScreenStack (created by NativeStackViewInner)
    in NativeStackViewInner (created by NativeStackView)
    in RCTView (created by View)
    in View (created by SafeAreaInsetsContext)
    in SafeAreaProviderCompat (created by NativeStackView)
    in NativeStackView (created by NativeStackNavigator)
    in NativeStackNavigator (created by RootNavigator)
    in RootNavigator (created by Navigation)
    in EnsureSingleNavigator
    in BaseNavigationContainer
    in ThemeProvider
    in NavigationContainerInner (created by Navigation)
    in Navigation (created by App)
    in RNCSafeAreaProvider (created by SafeAreaProvider)
    in SafeAreaProvider (created by App)
    in App (created by ExpoRoot)
    in ExpoRoot
    in RCTView (created by View)
    in View (created by AppContainer)
    in RCTView (created by View)
    in View (created by AppContainer)
    in AppContainer
at node_modules/react-native/Libraries/Core/ExceptionsManager.js:104:6 in reportException
at node_modules/react-native/Libraries/Core/ExceptionsManager.js:172:19 in handleException
at node_modules/react-native/Libraries/Core/ReactFiberErrorDialog.js:43:2 in showErrorDialog
at node_modules/@react-navigation/native/src/useThenable.tsx:25:13 in resolve
```

</details>

**Full Repo**: https://github.com/brianbancroft/issue-init-react-native-ble-plx

In this example, the crashing code can be found in `App.tsx`. 

## Motivation

I saw your library among one or two others in the RN world. I am still new when it comes to mobile development, and chose your library because it appears you have a team of motivated maintainers who are continuously making changes to the app big and small. That's really nice. 

I hope that in solving my problem, I can help through either observing a weakness in an otherwise robust platform should this be a real thing. In the more likely case of a mistake which I made, I hope to help edit the documentation just a little so that someone in a similar position as to myself doesn't make the same mistake. 

Thanks for any thoughts/help in advance.

mciechanowicz, 2022-01-31:
Hi.

Do you detach the expo project? [Configuration & Installation - Expo](https://github.com/dotintent/react-native-ble-plx#expo)

Do you follow the instructions for android: [android installation](https://github.com/dotintent/react-native-ble-plx#android-example-setup) ?

brianbancroft, 2022-01-31:
No I did not. Just came to that realization. 

Closing this issue.

closed 2022-01-31 by brianbancroft

gabefgonc, 2023-03-01:
getting the same error, but followed all the steps for expo and android installation.

kaushik1094, 2023-05-12:
Same error. followed all the steps