dotintent/react-native-ble-plx #955: Is react-native-ble-plx supporting auto linking, if not what are the steps for manual linking?

adityareddyav, 2022-02-04:
Hi 

-  I am running the latest version reacr-native-ble-plx 2.0.3
- I checked the documentation and found no answer
- I checked to make sure that this issue has not already been filed
- I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. Question has been already asked on StackOverflow but went unanswered - https://stackoverflow.com/questions/63787443/js-react-native-how-can-i-solve-this-error-for-building-react-native-app-for-i

## Question

- Is react-native-ble-plx supporting auto linking?
- If it is not supporting what are the steps for manual linking?

Matheuschn, 2022-02-19:
I'm pretty sure this library supports auto-linking. To solve your error, try unlinking the library as suggested in the message and running `npm install`. This should automatically link the library.

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska