dotintent/react-native-ble-plx #957: Android force kill doesn't disconnect

fgrs, 2022-02-16:
I know this was answered before and some suggested to use AppState/destroy solution but in my case I need characteristic data exchange also when app is in a background.

the actual issue occurs when my BT device uses fast mode and sends characteristics approx every 1 sec then if I swipe/kill the app the connection remains and from this point I can only power cycle BT device. If in normal mode (data approx every 3 sec) everything works fine and device disconnects perfectly.

I also noticed that leaving transaction id empty for monitorCharacteristicForService prevents BT device from disconnecting in both fast and normal mode which got me thinking that the initial issue is somehow related transaction id.

Does anybody know how to get around this problem?

Matheuschn, 2022-02-19:
Does this bug affect Android, iOS or both?

fgrs, 2022-02-19:
I can confirm it only affects Android devices

scsonic, 2022-03-22:
Too many android phone didn't implement BLE stack like Pixel/nexus phones.

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska