dotintent/react-native-ble-plx #958: bitrise android build - Task :react-native-ble-plx:processDebugAndroidTestManifest FAILED

ProPanek, 2022-02-28:
## Expected Behavior

In bitrise the android app should build successfuly.

## Current Behavior

Build is failing due to the minSdk compatibility.

### Steps to Reproduce

After some time while building the app for the android. I'm getting this error message.

```
> Task :react-native-ble-plx:processDebugAndroidTestManifest FAILED
[androidx.vectordrawable:vectordrawable-animated:1.0.0] /Users/vagrant/.gradle/caches/transforms-3/5232169e73d4423ea21d51876a2d84b1/transformed/vectordrawable-animated-1.0.0/AndroidManifest.xml Warning:
	Package name 'androidx.vectordrawable' used in: androidx.vectordrawable:vectordrawable-animated:1.0.0, androidx.vectordrawable:vectordrawable:1.0.1.
/Users/vagrant/git/node_modules/react-native-ble-plx/android/build/intermediates/tmp/manifest/androidTest/debug/manifestMerger4152730066210792536.xml:5:5-74 Error:
	uses-sdk:minSdkVersion 18 cannot be smaller than version 21 declared in library [com.facebook.react:react-native:0.67.2] /Users/vagrant/.gradle/caches/transforms-3/e6d0898a74320d34d15c5decc8322fa1/transformed/jetified-react-native-0.67.2/AndroidManifest.xml as the library might be using APIs not available in 18
	Suggestion: use a compatible library with a minSdk of at most 18,
		or increase this project's minSdk version to at least 21,
		or use tools:overrideLibrary="com.facebook.react" to force usage (may lead to runtime failures)
```

And in `android/build.grade` I have this config

```
buildscript {
    ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 21
        compileSdkVersion = 30
        targetSdkVersion = 30
        ndkVersion = "21.4.7075529"
    }
    ...
}
```

### Context

Before adding ble-plx the app was building successfully. The sdk is installing properly on the bitrise workflow, building is failing every time in the same spot that is mentioned above. Any help would be appreciated.

ProPanek, 2022-02-28:
I've found this PR https://github.com/dotintent/react-native-ble-plx/pull/925 that might fix the issue, when can we have it merged?

ablack34, 2023-02-21:
Hey, facing this problem also. Any solution?

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska