dotintent/react-native-ble-plx #960: Question: Does this library support connection to external QR scanner and external camera?

karel-suchomel-edgy, 2022-03-12:
Hello,
I am trying to create an RN app that connects to an external QR scanner that scans barcodes and sends data to RN and an external camera for taking pictures and displaying them in RN.
I came across this package but haven't found any implementation of this feature anywhere so I'm just curious if it is possible and if anybody has experience with it and could provide any sort of help as I'm kinda new to RN.

peeterkask, 2022-03-16:
Yes, it does. Recently implemented one QR/barcode scanner and it worked as expected. Just need to make sure that the scanner device supports BLE and is in BLE mode.

The quick start guide can be followed to connect to the device: https://github.com/dotintent/react-native-ble-plx/blob/master/INTRO.md
When BLE connection is established correct service and characteristic needs to be found using service discovery: https://github.com/dotintent/react-native-ble-plx/wiki/Device-Service-Discovery
Corresponding characteristic can be monitored for data input: https://github.com/dotintent/react-native-ble-plx/wiki/Characteristic-Notifying

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-10-06 by abursztynowska