dotintent/react-native-ble-plx #965: React Native 0.66

Irenedan, 2022-04-13:
Hi Mciechanowicz：
Thank you so much for all of these!

I noticed that there are already commits related to the react-native upgrade, so how is the progress now and when is the merge expected to be completed?

React Native & React version upgrade #925  @mciechanowicz

hugows, 2022-05-12:
React Native is on version 0.68 now - anyone knows the latest version that works well? I don't have major problems with my apps on older verrsions, should probably keep those.

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-10-06 by abursztynowska