dotintent/react-native-ble-plx #966: startDeviceScan error - Cannot start scanning operation - error code: 600

bear1030, 2022-04-13:
I have the app built with this library.

On Galaxy S10 startDeviceScan method returns the following error.

`{"message":"Cannot start scanning operation","errorCode":600,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":"Bluetooth cannot start (code 0)","name":"BleError"}`

- I am running the latest version
- I checked the documentation and found no answer
- I checked to make sure that this issue has not already been filed and found  #738 and #390 , but couldn't find an answer.

DreamGamer, 2022-04-28:
Hi, i had the same issue with my Samsung S10 and got it fixed. The Samsung S10 has Android 12 and this android update has new permissions. So i updated my AndroidManifest.xml file with these Permissions:
```
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE"/>
<uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
<uses-feature android:name="android.hardware.bluetooth" android:required="true"/>
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
```
And i also use this package to manage the Persmissions: [https://github.com/zoontek/react-native-permissions](https://github.com/zoontek/react-native-permissions)

groopmisha, 2022-05-10:
Hi, I had the same issue on Android 12, on Android 11 everything worked fine. Thanks, @DreamGamer, I tried react-native-permissions with the list above and everything ok

bear1030, 2022-05-25:
@DreamGamer 

I added all those permissions to AndroidManifest.xml file.

Which permissions do you check with https://github.com/zoontek/react-native-permissions?

Thank you!

bear1030, 2022-05-25:
@DreamGamer  I resolved it. I had to request the 'BLUETOOTH_SCAN' and 'BLUETOOTH_CONNECT'.
You saved my day.
Thank you again.

ShaoGongBra, 2022-07-02:
@bear1030 How did you ask for permission? Can you share the code? I added the permission, thank you, it's still the same

bear1030, 2022-07-02:
`import {PERMISSIONS, RESULTS, requestMultiple} from 'react-native-permissions';
...
  await requestMultiple([
    PERMISSIONS.ANDROID.BLUETOOTH_SCAN,
    PERMISSIONS.ANDROID.BLUETOOTH_CONNECT,
  ])
`

ShaoGongBra, 2022-07-02:
@bear1030 Don't need this permission: android.permission.ACCESS_FINE_LOCATION

bear1030, 2022-07-02:
@ShaoGongBra  That's what you need to grant for this package as a default.
Here is my full code.

`import {PERMISSIONS, RESULTS, requestMultiple} from 'react-native-permissions';`
...
`await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION);`
`await requestMultiple([`
`PERMISSIONS.ANDROID.BLUETOOTH_SCAN,`
`PERMISSIONS.ANDROID.BLUETOOTH_CONNECT,`
`]);`

closed 2023-09-27 by dominik-czupryna-withintent