dotintent/react-native-ble-plx #967: Could not determine artifacts for com.github.Polidea:MultiPlatformBleAdapter:b0272c0fae040cc90f790481c54e8f2d45304339: Skipped due to earlier error

stevecode21, 2022-04-14:
## Prerequisites

- [ x] I am running the latest version
- [x ] I checked the documentation and found no answer
- [ x] I checked to make sure that this issue has not already been filed

## Expected Behavior

Android build should finish successfully

## Current Behavior

When I try to generate an APK (Build) in **AppCenter** for android with the react-native-ble-plx package installed, I get the error: `Could not determine artifacts for com.github.Polidea:MultiPlatformBleAdapter:b0272c0fae040cc90f790481c54e8f2d45304339: Skipped due to earlier error`
### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

You can simply try to generate a build locally or in AppCenter.

### Context

I'm currently using 2.0.3 version, but seems like in this version the library is using a specific commit from the MultiPlatformBleAdapter repository instead of its final version.

* Library version: 2.0.3
* Platform: Android.

asagalovskiyLB, 2022-04-20:
Same issue; tried pointing the dependency to https://www.jitpack.io/com/github/dotintent/MultiPlatformBleAdapter/0.1.9/MultiPlatformBleAdapter-0.1.9.pom
...to no avail

reinhardholl, 2022-05-03:
Same issue here. Have not been able to do a clean build for some time..

rrvilo, 2022-05-12:
I was facing this similar error as well for a long time, until I realised that I was using exactly the same version of this library in another project, which works. I got this one fixed by changing the order of the repositories in the project level `build.gradle`. I have no deep-level understanding why this happens, but I think the build tries to install the `com.github.Polidea:MultiPlatformBleAdapter:b0272c0fae040cc90f790481c54e8f2d45304339` from a repository where it doesn't exist, instead of installing it from `'https://www.jitpack.io'`.

Also according to the gradle documentation, the order of the repositories does matter https://docs.gradle.org/current/userguide/declaring_repositories.html#sec:declaring_multiple_repositories

This is my current working setup:

```
// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    ext {
        buildToolsVersion = "30.0.0"
        minSdkVersion = 22
        compileSdkVersion = 30
        targetSdkVersion = 30
    }
    repositories {
        google()
        mavenCentral()
        jcenter()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:4.2.2")
        classpath("com.google.gms:google-services:4.3.4")
    }
}

allprojects {
    repositories {
        mavenCentral()
        mavenLocal()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }
        google()
        jcenter()
        maven { url 'https://www.jitpack.io' }
    }
}
```

dominik-czupryna-withintent, 2023-09-29:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs. Please confirm if the issue still persists and create a new issue if it still does.

closed 2023-09-29 by dominik-czupryna-withintent