dotintent/react-native-ble-plx #970: How to obtain data from multiple manufacturers?

luhaimin, 2022-04-22:
Hi, I can receive two manufacturers' data in NRF app,
Why can't I get two broadcast data when I use startDeviceScan in react-native-ble-plx?
I want to know, how can I get the first broadcast data?

[image]

[image]

Very urgent, looking forward to your answer

wllamb, 2022-12-19:
Also having this issue, nRF Connect shows two different values for manufacturer data. I was able to get the correct one for awhile and all of sudden it's now giving me the opposite. Would be ideal if I could see both

intent-kacper-cyranowski, 2024-06-19:
Hi @luhaimin and @wllamb 
Did `rawScanRecord` function solve your issues?

intent-kacper-cyranowski, 2024-06-25:
I’m going to close the issue due to inactivity, in case of any problems feel free to open a new issue

closed 2024-06-25 by intent-kacper-cyranowski