dotintent/react-native-ble-plx #972: manager.discoverAllServicesAndCharacteristicsForDevice(item.id) error from device

JamesOwen7, 2022-04-23:
I'm receiving the following error when connecting attmepting to connect to a glucose monitor. Does error 101 suggest that my device does not use bluetooth ble and I should in fact be attempting via bluetooth classic?

I'm trying to implement the function below:
  const connected = (item) => {
    manager.connectToDevice(item.id)
    .then(() => {
      
      manager.discoverAllServicesAndCharacteristicsForDevice(item.id)
      .then((results) => {console.log('results:'); console.log(results)})
      
    }); 
    setIsConnected(item.id);
  }

{"_manager": {"_activePromises": {}, "_activeSubscriptions": {"1": [Object]}, "_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", "103": "BluetoothLE is in unknown state", "104": "BluetoothLE is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": "Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", "204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service {serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": "Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", "404": "Characteristic {characteristicUUID} not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: {internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {}, "_scanEventSubscription": {"remove": [Function remove]}, "_uniqueId": 5}, "id": "04:52:C7:FB:01:1B", "isConnectable": null, "localName": null, "manufacturerData": null, "mtu": 23, "name": "LE-Bose Revolve SoundLink", "overflowServiceUUIDs": null, "rssi": null, "serviceData": null, "serviceUUIDs": null, "solicitedServiceUUIDs": null, "txPowerLevel": null}

JamesOwen7, 2022-04-23:
Or can anyone suggest how I might fix this if in fact my device is compatible

JamesOwen7, 2022-04-23:
PS - I believe the app is connected to the device. As the device has a blue flashing light which indicates it's available to be connected to. It turns solid blue when connected to. This process does happen

Tadejo9, 2022-04-23:
On what platform does the error occur? And what version of react native, library and os are you using?
And can you check manufacturer website of the device, to verify if ble is an option for it.

JamesOwen7, 2022-04-23:
I'm using windows - trying to get this to run on android.
RN Version: 0.68.1

I can't find any info on manufacturers website. I was going to call them but they are shut today

JamesOwen7, 2022-04-25:
I've managed to connect and retrieve the characteristics :)

Tadejo9, 2022-04-25:
Nice :). What was the problem?

JamesOwen7, 2022-04-25:
Honestly...I think just the implementation of the methods. I found a blog which talked through setting up and connecting to a device, and now it works! Now to read the data...I'm sure I'll be posting another question in the near future

JamesOwen7, 2022-04-25:
@Tadejo9 are you able to help with this? Trying to monitor data from the device. I get an error which says the below:
```
TypeError: undefined is not a function (near '...device.monitorCharacteristicForService('00001808-0000-1000-8000-00805f9b34fb', '00002A18-0000-1000-8000-00805f9b34fb').then...')
```
```
const connectDevice = device => {
    manager.stopDeviceScan();
  manager.connectToDevice(device.id).then(async device => {
               await device.discoverAllServicesAndCharacteristics();
               manager.stopDeviceScan();
               setDisplayText(`Device connected\n with ${device.name}`);
                setConnectedDevice(device);
                     setDevices([]);
      device.monitorCharacteristicForService('00001808-0000-1000-8000-00805f9b34fb', '00002A18-0000-1000-8000-00805f9b34fb').then(async service => {
        console.log(service)   
      
    });
```

closed 2022-04-25 by JamesOwen7