dotintent/react-native-ble-plx #973: TypeError: undefined is not a function (near '...device.monitorCharacteristicForService...')

JamesOwen7, 2022-04-25:
I'm trying to pull data from a glucose monitor. I've connected successfully and have the correct UUIDs for the relevant services and characteristics but I'm getting the error below:

```
TypeError: undefined is not a function (near '...device.monitorCharacteristicForService...')
```
Does anyone know why this might be? Code below shows the connecting function and the attempt to monitor the data. Both are being i nitiated by button presses.

Note I'm trying to run on android.

```
  const connectDevice = device => {
    manager.stopDeviceScan();
  manager.connectToDevice(device.id).then(async device => {
              await device.discoverAllServicesAndCharacteristics();
              manager.stopDeviceScan();
              setDisplayText(`Device connected\n with ${device.name}`);
              setConnectedDevice(device);
              setDevices([]);

              device.services().then(async service => {
                console.log(service)
              })

  }
  );
  }

  const getData = async device => { await
    device.monitorCharacteristicForService(
      '00001808-0000-1000-8000-00805f9b34fb', 
      '00002A18-0000-1000-8000-00805f9b34fb', 
      (error, characteristic) =>{
        if (error) {
          console.error("Error at receiving data from device", error);
          return;
        } else {
          console.log('Characteristic', characteristic)
        }},
  )};
```

utkarshpandey12, 2022-09-16:
same issue on calling device. write_characteristics_without_response()
anyone ?
@JamesOwen7 
@corymsmith 
@below 
@Kaspa 
@bitcrumb

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska