dotintent/react-native-ble-plx #975: Can't read characteristics from HRM

JamesOwen7, 2022-04-30:
Can anyone help with the below. I'm trying to read a random characteristic from a heart rate monitor before I try and actually pull the heart rate data. I' getting the error:

```
Error: Exception in HostFunction: Malformed calls from JS: field sizes are different.

[[4,29],[21,0],[[61,2000,1651308190120,false]],168]
```
The code I am using is as follows:

```
const connectDevice = device => {
    // manager.stopDeviceScan();
    //console.log(device)
  manager.connectToDevice(device.id).then(async device => {
              await device.discoverAllServicesAndCharacteristics()
              .then((results) => { 
                console.log(results);
                // manager.servicesForDevice(device.id)
                //   .then((services) => { 
                //     console.log('services');
                    //console.log(services);
                    //console.log(this.getValueFromBase64("AwYA5AcEDRAmCgAAeLDx"));
                    //console.log(this.getValueFromBase64("AwUA5AcEBxYRFwAAW7Dx"));
                    manager.readCharacteristicForDevice(
                      device.id, 
                      '0000180d-0000-1000-8000-00805f9b34fb',
                      '00002A38-0000-1000-8000-00805f9b34fb', 
                      (error, characteristic) => {  
                        console.log(characteristic.value);
                        if (error) {
                              console.log('Error: ' + error.message + '. My error message')
                            
                          }
                          if (true) {
                            console.log('Getting value')
                            console.log(characteristic.value)
                          }                  
                          //console.log(response, 'response')
                          console.log('hello')
                        }
                      )
                  //});
              });
              // console.log(devices)
              manager.stopDeviceScan();
              setDisplayText(`Device connected\n with ${device.name}`);
              setConnectedDevice(device);
              setDevices(device);
              // console.log(device)
              device.services().then(async service => {
                // console.log(service)
              })

  })}
```
Does anyone know why I can't read the data? Is there something very wrong with my function...?

Thanks in advance!

Saikrishna1834, 2022-11-16:
@JamesOwen7 were you able to slove this issue, I am also facing the same issue. getting same error when trying to read data from Smart Weight Scale

little-steps, 2023-02-23:
i am also facing this issue is it solved?

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska