dotintent/react-native-ble-plx #977: How to interpret battery level

ibrahim-delice, 2022-05-12:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ x] I am running the latest version
- [ x] I checked the documentation and found no answer
- [ x] I checked to make sure that this issue has not already been filed
- [ x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Please describe the question.

I am reading the battery level from a device but do not know how to interpret the battery level percentages from the response.

This is how I read the characteristic:
```
const batteryChar = await device.readCharacteristicForService(
          '0000180F-0000-1000-8000-00805f9b34fb',
          '00002a19-0000-1000-8000-00805f9b34fb',
        );
```

Response:
```
{
  "id": 10789092832,
  "uuid": "00002a19-0000-1000-8000-00805f9b34fb",
  "isNotifiable": true,
  "isNotifying": true,
  "isReadable": true,
  "isWritableWithResponse": false,
  "serviceID": 10738251328,
  "value": "ZA==",
  "isIndicatable": false,
  "serviceUUID": "0000180f-0000-1000-8000-00805f9b34fb",
  "deviceID": "xxxxx-xxxx-xxxx-xxxx-xxxx",
  "isWritableWithoutResponse": false,
  [...]
}
```

The only property that looks like could be the answer is the `value` property, the value itself looks different.

How do I interpret the battery level from the response?

ibrahim-delice, 2022-05-19:
> 

Thank you but I managed to do it like this:

```typescript
const batteryCharacteristic = await device.readCharacteristicForService(
      '0000180F-0000-1000-8000-00805f9b34fb',
      '00002a19-0000-1000-8000-00805f9b34fb',
    );

    const currentBatteryLevel = Buffer.from(
      batteryCharacteristic && batteryCharacteristic.value
        ? batteryCharacteristic.value
        : '',
    ).readInt8();
```

Chasty, 2022-09-16:
@ibrahim-delice I get this value `Nw==` and then want to use your method and this error arises.

`RangeError: Trying to access beyond buffer length, js engine: hermes`

any idea?

Sikandarkhan, 2022-12-08:
I faced similar issue. Managed to solve from the documentation link . 
https://github.com/dotintent/react-native-ble-plx/wiki/=--FAQ:-Passing-And-Retrieving-Of-Characteristic-Value#reading-value-from-a-characteristic 

hope you find this useful.

closed 2023-09-27 by dominik-czupryna-withintent