dotintent/react-native-ble-plx #980: writeCharacteristicWithoutResponse time between each message

gregogalante, 2022-05-18:
Hello,

I'm sending a large file (100kb) to a peripheral with MTU set to 20 bytes using writeCharacteristicWithoutResponse function. My code is a for loop that call writeCharacteristicWithoutResponseForDevice function at every cycle for a single packet of 20 bytes.

I notice the messages arrives to the peripheral with a strange interval:
4-5 packets come with 1ms of difference with the previous
1 packets come with 40-50ms of difference with the previous

I have tested the problem with a BLE Sniffer on the peripheral and with a sample BLE client on Mac (using node.js and bleno) but i get the same problem.

## Expected Behavior

I expect to receive each packet with a similar time interval.

## Current Behavior

I receive some packets with a small interval (< 1ms) and some with a really large interval (30-40ms)

### Steps to Reproduce

React native code (running on Android 12 with the latest library version)

```javascript
const packetsLength = 5773
const packes = [.....] // array of arrays of 20 bytes
for (let i = 0; i < packetsLength; i++) {
      await BleManager.writeCharacteristicWithoutResponseForDevice(
        deviceId,
        SERVICE_UUID,
        CHARWRITE_UUID,
        bytesToMessage(packets[i])
      )
}
```

Bleno BLE peripheral to check time between messages

```javascript
const bleno = require('bleno')
const BlenoPrimaryService = bleno.PrimaryService
const BlenoCharacteristic = bleno.Characteristic

const SERVICE_UUID = 'd973f2e0-b19e-11e2-9e96-0800200c9a66'
const CHARWRITE_UUID = 'd973f2e2-b19e-11e2-9e96-0800200c9a66'
const CHARREAD_UUID = 'd973f2e1-b19e-11e2-9e96-0800200c9a66'

/**
 * Characteristic for writing
 */

let lastMessageTime = 0

const characteristic = new BlenoCharacteristic({
  uuid: CHARWRITE_UUID, // or 'fff1' for 16-bit
  properties: ['read', 'write', 'writeWithoutResponse', 'notify', 'indicate'],
  onWriteRequest: () => {
    const currentTime = new Date().getTime()
    if (!lastMessageTime) lastMessageTime = currentTime
    const timeDiff = currentTime - lastMessageTime
    console.log('write request', timeDiff)
    lastMessageTime = currentTime
  }
})

/**
 * Main
 */

bleno.on('stateChange', function(state) {
  console.log('BLENO -> stateChange: ' + state)

  if (state === 'poweredOn') {
    bleno.startAdvertising('echo', [SERVICE_UUID])
  } else {
    bleno.stopAdvertising()
  }
})

bleno.on('advertisingStart', function(error) {
  console.log('BLENO -> advertisingStart: ' + (error ? 'error ' + error : 'success'))

  if (!error) {
    bleno.setServices([
      new BlenoPrimaryService({
        uuid: SERVICE_UUID,
        characteristics: [characteristic]
      })
    ])
  }
})
```

Current Bleno output

```shell
write request 42
write request 1
write request 0
write request 1
write request 1
write request 42
write request 0
write request 1
write request 1
write request 0
write request 1
write request 1
write request 41
write request 1
write request 0
write request 1
write request 43
write request 1
write request 0
write request 1
write request 0
write request 43
.....
```

Ble Sniffer example on real peripheral (the time between every message is in the second column)

[image]

### Context

* Library version: 2.0.3
* Platform: Android

```json
{
  "name": "prototype",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "@react-native-picker/picker": "^2.4.1",
    "@react-native-voice/voice": "^3.2.3",
    "@react-navigation/native": "^6.0.10",
    "@react-navigation/native-stack": "^6.6.1",
    "@sayem314/react-native-keep-awake": "^1.1.0",
    "prop-types": "^15.8.1",
    "react": "17.0.2",
    "react-native": "0.68.0",
    "react-native-ble-plx": "^2.0.3",
    "react-native-picker-select": "^8.0.4",
    "react-native-safe-area-context": "^4.2.4",
    "react-native-screens": "^3.13.1",
    "react-native-webview": "^11.18.1",
    "react-native-wifi-reborn": "^4.5.0",
    "rn-fetch-blob": "^0.12.0"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@babel/runtime": "^7.12.5",
    "@react-native-community/eslint-config": "^2.0.0",
    "babel-jest": "^26.6.3",
    "eslint": "^7.32.0",
    "jest": "^26.6.3",
    "metro-react-native-babel-preset": "^0.67.0",
    "react-test-renderer": "17.0.2"
  },
  "jest": {
    "preset": "react-native"
  }
}
```

gregogalante, 2022-05-19:
Looking at the sniffer table i see that there are exactly 43ms between every timeout.

KingOfTNT10, 2022-05-31:
Hi @gregogalante!
I'm very new to this library and was wondering, how did you get the SERVICE_UUID? I can't seem to find a way except for device.services which return null. Did you do it this way or am I missing something? does the BLE device matter?

btw sorry for asking you in your own issue but I hope it's ok. I just need an answer quickly so that's the only thing I could think of. So sorry again for not answering it.

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska