dotintent/react-native-ble-plx #984: Bug Report - Can't create BLE manager

KingOfTNT10, 2022-05-29:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x ] I am running the latest version
- [x ] I checked the documentation and found no answer
- [x ] I checked to make sure that this issue has not already been filed

## Expected Behavior

I expect it to work like is shown here: https://github.com/dotintent/react-native-ble-plx/blob/master/INTRO.md

## Current Behavior

the BLE manager doesnt work and when I console.log it it shows this:

```
{"_activePromises": {}, "_activeSubscriptions": {}, "_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a 
bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is 
not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", "103": "BluetoothLE is in unknown state", "104": "BluetoothLE 
is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": 
"Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", 
"204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU 
size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for 
device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not 
discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service 
{serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": 
"Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic 
{characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", "404": "Characteristic {characteristicUUID} 
not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to 
characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: 
{internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic 
{characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic 
{characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic 
{characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service 
{serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: 
{internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as 
well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {}, "_scanEventSubscription":
 {"remove": [Function remove]}, "_uniqueId": 0}
```

what should I do?
### Steps to Reproduce

followed until the third step: https://github.com/dotintent/react-native-ble-plx/blob/master/INTRO.md then when I noticed it doesn't work I stopped so it's actually the first step.

### Context

* Library version: 2.0.3
* Platform: Android.
* Platform logs (logcat/XCode): No output
* JS logs: No output
* Contents of the `package.json` file: 
```
{
  "name": "on_air_project",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "@react-native-async-storage/async-storage": "^1.17.5",
    "@react-native-community/masked-view": "^0.1.11",
    "@react-navigation/native": "^6.0.10",
    "@react-navigation/stack": "^6.2.1",
    "react": "17.0.2",
    "react-native": "0.68.2",
    "react-native-ble-manager": "^8.1.0",
    "react-native-ble-plx": "^2.0.3",
    "react-native-fs": "^2.20.0",
    "react-native-gesture-handler": "^2.4.2",
    "react-native-reanimated": "^2.8.0",
    "react-native-safe-area-context": "^4.2.5",
    "react-native-screens": "^3.13.1",
    "react-navigation": "^4.4.4",
    "rn-nodeify": "^10.3.0"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@babel/runtime": "^7.12.5",
    "@react-native-community/eslint-config": "^2.0.0",
    "babel-jest": "^26.6.3",
    "eslint": "^7.32.0",
    "jest": "^26.6.3",
    "metro-react-native-babel-preset": "^0.67.0",
    "react-test-renderer": "17.0.2"
  },
  "jest": {
    "preset": "react-native"
  }
}
```

* Formatted code sample or link to a repository:

```js
import {BleManager} from 'react-native-ble-plx';
const manager = new BleManager();

  console.log(manager);
```

closed 2022-05-31 by KingOfTNT10

thejager, 2022-09-23:
How did you solve this? @KingOfTNT10

KingOfTNT10, 2022-09-23:
@thejager Hi, so thats actually the manager. It just stores all error codes inside the object.
If you need more help I might be able to help because the support isnt amazing here.