dotintent/react-native-ble-plx #988: ❌ ERROR: Invariant Violation: `new NativeEventEmitter()` requires a non-null argument. - Project Setup

nurcinozer, 2022-06-09:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I checked all example applications mentioned in README file and they don't work for me.
- [x] I followed every installation step mentioned in README file.

## Setup problem

I am trying to setup react-native-ble-plx to my expo app but I am getting this error:
` ERROR  Invariant Violation: `new NativeEventEmitter()` requires a non-null argument.`

### Steps to Reproduce

1. `import React, { useState } from "react";
import {
  View,
  Text,
  TouchableOpacity,
  StyleSheet,
  Button,
  ImageBackground,
  Image,
} from "react-native";
import { COLORS, SIZES, assets, styles1, styles2 } from "../constants";
import { Avatar } from "@rneui/themed";
import { BleManager } from "react-native-ble-plx";
import DeviceCard from "../components/DeviceCard";

const Home = ({ navigation }) => {
  const manager = React.useRef();
  manager.current = new BleManager({
    restoreStateIdentifier: 'com.baucapstone.smartdoorlock',
    restoreStateFunction: () => {},
    allowDuplicate: false,
  });

  var devices = [];

  const [isLoading, setIsLoading] = useState(false);

  const scanDevices = () => {
    setIsLoading(true);

    manager.startDeviceScan(null, null, (error, scannedDevice) => {
      if (error) {
        console.warn(error);
      }
      if (scannedDevice && !devices.find((dev) => dev.id === scannedDevice.id)) {
        console.log(scannedDevice)
        devices = [
          ...devices,
          {id: scannedDevice.id, name: scannedDevice.name},
        ];
      }
    });

    setTimeout(() => {
      manager.stopDeviceScan();
      setIsLoading(false);
    }, 5000);
  };

  return (
    <View
      style={{
        flex: 1,
      }}
    >
      <Button color="#3b5998" title="Scan bluetooth devices" onPress={scanDevices} />
      {devices.map((device) => <DeviceCard device={device} />)}
      <ImageBackground
        source={require("../assets/images/background.png")}
        resizeMode="cover"
        style={{
          flex: 1,
          justifyContent: "center",
        }}
      >
        <View
          style={{
            flex: 1,
            justifyContent: "space-around",
            alignItems: "center",
          }}
        >
          <View
            style={{
              width: "100%",
              alignItems: "flex-start",
              top: -45,
              left: 10,
            }}
          >
            <Text
              style={{
                fontSize: SIZES.extraLarge,
                fontWeight: "bold",
              }}
            >
              Home
            </Text>
          </View>
          <View>
            <TouchableOpacity
              style={{
                borderWidth: 2,
                borderColor: "#0782F9",
                width: 200,
                padding: 15,
                paddingRight: 30,
                borderRadius: 10,
                alignItems: "center",
                flexDirection: "row",
                justifyContent: "space-around",
                alignItems: "center",
              }}
            >
              <Image source={require("../assets/icons/unlock.png")} />
              <Text
                style={{
                  fontSize: 18,
                  fontWeight: "bold",
                }}
              >
                Lock
              </Text>
            </TouchableOpacity>
          </View>
          <View
            style={{ flexDirection: "row", alignItems: "center", top: -60 }}
          >
            <Avatar
              size="medium"
              rounded
              source={{ uri: "https://i.pravatar.cc/400" }}
            />
            <View
              style={{
                padding: 10,
              }}
            >
              <Text style={{ fontWeight: "bold" }}>
                last access 7.46AM by bilgesu
              </Text>
              <Text style={{ fontWeight: "bold" }}>door was unlocked</Text>
            </View>
          </View>
        </View>
      </ImageBackground>
    </View>
  );
};

export default Home;
`

* Library version: latest version
* Platform: iOS.

eddyclock, 2022-06-10:
Hi Nurcinozer,
i have also encountered this error and just find workaround solution as below in **index.js**
```
import {LogBox} from "react-native";
LogBox.ignoreLogs(['new NativeEventEmitter']); // Ignore log notification by message
LogBox.ignoreAllLogs(); //Ignore all log notifications
Root problem is not yet solved but still work
```
Eddy

diegobonagurio, 2022-06-16:
Are you using expo?

thgh, 2022-06-21:
I'm in a fresh Expo project and getting this slightly longer error:
```
`new NativeEventEmitter()` was called with a non-null argument without the required `addListener` method.
at node_modules/react-native/Libraries/EventEmitter/NativeEventEmitter.js:62:8 in constructor
at node_modules/react-native-ble-plx/src/BleManager.js:66:25 in constructor
```

I changed BleManager.js like this, but this is not a proper solution:
```diff
  /**
   * Creates an instance of {@link BleManager}.
   */
  constructor(options: BleManagerOptions = {}) {
+    if (!BleModule.addListener)BleModule.addListener=console.warn
+    if (!BleModule.removeListeners)BleModule.removeListeners=console.warn
```

vid553, 2022-07-15:
Having the same issue on a non-expo project with RN 0.67.4.
Workaround by @eddyclock works nicely.

nissaba, 2022-07-28:
having same problem, on both expo or expo ejected project. 
all I need to to is add this to my App.js 
import { BleManager } from "react-native-ble-plx";

const _BleManager = new BleManager();

on a bran new project

marchingband, 2022-08-01:
I have the same problem, using EXPO on a new project, on android.
It triggers a yellow warning, and the BLE never powers on.

caseyyee, 2022-10-12:
I'm also encountering the same issue using Expo.

Project seems dead, there hasn't been any commits for 10 months. 😮

correadevuk, 2022-12-19:
Expo users:
Upgraded @config-plugins/react-native-ble-plx to version 5.0.0, and the warning went away.

macksal, 2023-04-18:
I still get this issue on the latest @config-plugins/react-native-ble-plx, expo and react-native-ble-plx.

It seems the issue is as follows. NativeEventEmitter takes the nativeModule argument passed to the constructor so it can notify a native module of how many listeners there are (something to do with conserving bridge traffic, I suppose - if there are zero listeners on an emitter, the native layer can choose not to emit an event?). This is required on iOS, not entirely sure why.
- 1 (on iOS) nativeModule must be present and must include the addListener/removeListeners methods.
- 2  (on Android) nativeModule argument can be missing (2a), be passed and include the addListener/removeListeners methods (2b), or be passed without those methods (2c).

It seems the (2c) case issues this warning. It appears to be resolved by passing the nativeModule argument as `undefined` to NativeEventEmitter if `Platform.OS === "android"`.

Alternatively, we can add empty dummy implementations of the addListener/removeListeners methods when on Android. This can be done by monkey patching (in javascript) the BleModule exported from `react-native-ble-plx/src/BleModule` before creating a BleManager. This is a temporary hack and would better be done in the java layer.

I could PR either of these options, but not sure which is best and the repo seems to be unmaintained... I was initially tripped up by #730 thinking this warning was breaking my app on Android, but it was just the permissions issue.

bj97301, 2023-05-02:
I am getting this on "@config-plugins/react-native-ble-plx": "^6.0.0", and  "react-native": "0.71.7",

Has anyone fixed this in a pr yet? Android seems to not be working for me and this is the only feedback I am getting from the console.

bj97301, 2023-05-02:
> I still get this issue on the latest @config-plugins/react-native-ble-plx, expo and react-native-ble-plx.
>
> It seems the issue is as follows. NativeEventEmitter takes the nativeModule argument passed [...]

Any luck with your pr?

dominik-czupryna-withintent, 2023-09-29:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs. Please confirm if the issue still persists and create a new issue if it still does.

closed 2023-09-29 by dominik-czupryna-withintent

Quangdm-cdm, 2023-11-27:
Version 3.1.1 but still having this issue@@

JairajJangleSLIOT, 2024-02-08:
Still facing this issue on 3.1.2

JairajJangleSLIOT, 2024-02-08:
Solved this issue by deleting all Pods and rebuilding the project on iOS