dotintent/react-native-ble-plx #992: Bug Report: MTU prop on device does not seem to reflect correct data

da1z, 2022-06-16:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ y] I am running the latest version
- [ y] I checked the documentation and found no answer
- [ y] I checked to make sure that this issue has not already been filed

## Expected Behavior

Correct mtu returned

## Current Behavior

mtu value always 23

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.
I am testing on ios

1. connect to device
2. log device.mtu
3. you see 23

but I can send messages from peripheral large then 23, up to 500+ length without issues. Am I missing something?
i tried connect with requestMtu but it does not change behavior too
```

moooss, 2022-08-31:
Hi,
I'm encountering the same issue here.

**In fact the problem is:**

Whatever MTU I request, I'm getting 23.
But according to the real communication speed, the negociated MTU seems higher.

**The behavior is the following:** 

```
device.connect().then((d) => {
  console.log(d.mtu)
});
```
Result: `> 23`

```
device.connect({ requestMTU: 512 }).then((d) => {
  console.log(d.mtu)
});
```
Result: `> 23`

Android or iOS, same behavior.

But, I'm sending big files and : 
- with no requestMTU it's way faster on iOS, which confirms that [default mtus are not de same according to the platform](https://github.com/dotintent/react-native-ble-plx/wiki/MTU-Negotiation).
- with { requestMTU: 512 }, Android works faster than with no requestMTU, so the mtu negociated is higher than 23

dominik-czupryna-withintent, 2023-09-29:
- MTU by default is 23
- iOS devices negotiate automatically MTU value when you connect the device
- For Android devices, you have to use requestMTU command to change it

closed 2023-09-29 by dominik-czupryna-withintent