dotintent/react-native-ble-plx #994: React native link

Talha-raj, 2022-06-29:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I checked all example applications mentioned in README file and they don't work for me.
- [ ] I followed every installation step mentioned in README file.

## Setup problem

Please describe the problem you are experiencing.

PS E:\React Native\WebView> react-native link
error Unrecognized command "link".
info Run "react-native --help" to see a list of all available commands.
PS E:\React Native\WebView>

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. ...

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://dotintent.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: X.Y.Z
* Platform: Android/iOS.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:

jongbelegen, 2022-07-26:
React native link command is removed in [0.69](https://reactnative.dev/blog/2022/06/21/version-069#breaking-changes).
It should work without this command :-)

jaska120, 2022-08-23:
@jongbelegen A little offtopic, but have you managed to use this library past RN 0.66?