dotintent/react-native-ble-plx #996: Bug Report

savety6, 2022-07-04:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [1] I am running the latest version
- [1] I checked the documentation and found no answer
- [1] I checked to make sure that this issue has not already been filed

## Expected Behavior

I have an app that has to write a message into a specific service in both android and ios

## Current Behavior

The app is working perfectly fine in android but in ios gives back an error 

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Create a new project: expo init
2. Install the packages: expo add react-native-ble-plx @config-plugins/react-native-ble-plx expo-dev-client
3. Add @config-plugins/react-native-ble-plx to the expo.plugins array in your app.json
4. Build the native app (i used: eas build -p all --profile development)

### Context

I made an app that has to write a simple code to an ESP32 chip and in android it works properly but when i build it for ios the app returns me this error: BleError: Characteristic 0000abf1-0000-1000-8000-00805f9b34fb write failed for device B88BF908-9E6F-7219-67DC-1D14ACCF6CB3 and service 0000abf0-0000-1000-8000-00805f9b34fb
the ESP32 is the same and when I try to send the code from the same phone with a ble test app from app store it works so the problem is not in my device

* Library version: 2.0.3
* Platform: Android/iOS.
* JS logs:
BleError: Characteristic 0000abf1-0000-1000-8000-00805f9b34fb write failed for device B88BF908-9E6F-7219-67DC-1D14ACCF6CB3 and service 0000abf0-0000-1000-8000-00805f9b34fb
* Contents of the `package.json` file:
{
  "name": "bluwi",
  "version": "1.0.0",
  "scripts": {
    "start": "expo start --dev-client",
    "android": "expo run:android",
    "ios": "expo run:ios",
    "web": "expo start --web",
    "eject": "expo eject"
  },
  "dependencies": {
    "@config-plugins/react-native-ble-plx": "^2.0.0",
    "@react-native-async-storage/async-storage": "~1.17.3",
    "@react-navigation/native": "^6.0.10",
    "@react-navigation/native-stack": "^6.6.2",
    "expo": "~45.0.0",
    "expo-dev-client": "~1.0.0",
    "expo-splash-screen": "~0.15.1",
    "expo-status-bar": "~1.3.0",
    "react": "17.0.2",
    "react-dom": "17.0.2",
    "react-native": "0.68.2",
    "react-native-ble-plx": "^2.0.3",
    "react-native-safe-area-context": "4.2.4",
    "react-native-screens": "~3.11.1",
    "react-native-web": "0.17.7"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@types/react": "~17.0.21",
    "@types/react-native": "~0.67.6",
    "typescript": "~4.3.5"
  },
  "private": true
}

* Formatted code sample or link to a repository:

```js
async function sendInfo(value, name) {
        manager.writeCharacteristicWithResponseForDevice(
            device?.id,
            servUuid,
            writeUuid,
            base64.encode(value.toString()),
        ).then(characteristic => {
            console.log('send valuse  :', base64.decode(characteristic.value));

            const newState = state.filter((s)=>{
                return s?.empty != true
            })
            const value = {
                id: device?.id,
                code: code +'\r\n',
                key: name,
                empty: false,
            }
            setState([ ...newState, value])
            let refresh = true
            navigation.navigate('Home', {refresh})
        }).catch(error => {
            console.log(error);
        })
    }
```

roysG, 2022-07-22:
Same, Did you fix it?

savety6, 2022-07-29:
Yes , actually the problem was in me- I tried to write without responses on a characteristic with response.For some reason Android phones don't care whether you write on a characteristic with or without response.

closed 2022-07-29 by savety6