dotintent/react-native-ble-plx #1001: Using Expo on Android, ask permissions?

marchingband, 2022-08-07:
I am using EXPO, with the eas build plugin.
On iOS, the app requests permissions when it first opens.
On Android it does not ask permissions, users must manually give permissions from their device settings, both for Location and Bluetooth.
Is there a way to make Android ask for permissions like it does on iOS?
Thank you!!

josmas, 2022-09-02:
You can use PermissionsAndroid directly from RN https://reactnative.dev/docs/0.68/permissionsandroid or a community package such as https://www.npmjs.com/package/react-native-permissions

marchingband, 2022-09-02:
In EXPO I believe the plugin is responsible for asking permissions. Do the maintainers of this repo also maintain the expo plugin?

benjaminhera, 2023-06-19:
This behaviour is giving me a headache, because it instantly shows up the permission modal, which interferes with my splash screen on ios and makes my app unusable... On android I use PermissionsAndroid from react to do the logic manually.

Please have the option to disable automatic permission requesting for ios. Thanks.

marchingband, 2023-06-19:
@benjaminhera I think that is the native behavior, not something that can be controlled on iOS.

closed 2023-09-27 by dominik-czupryna-withintent