dotintent/react-native-ble-plx #1002: Feature Request would you add addListener and removeListeners to BleClientManager.java?

sunggun1, 2022-08-13:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed

## Current Behavior

 WARN  `new NativeEventEmitter()` was called with a non-null argument without the required `addListener` method.
 WARN  `new NativeEventEmitter()` was called with a non-null argument without the required `removeListeners` method.

## New behavior

No warnning 

### Acceptance criteria

Please provide detailed steps to accomplish proposed feature.

1. I npm install react-native-ble-plx --legacy-peer-deps 
2. "react-native-ble-plx": "^2.0.3",
3. I just added code below to BleClientManager.java

@ReactMethod
    public void addListener(String eventName) {
        // Keep: Required for RN built in Event Emitter Calls.
    }

    @ReactMethod
    public void removeListeners(Integer count) {
        // Keep: Required for RN built in Event Emitter Calls.
    }

### Breaking changes

Does a feature breaks compatibility with previous version? Probably no.

closed 2022-08-13 by sunggun1

pwellner, 2022-08-23:
How was this completed?

sunggun1, 2022-09-04:
> How was this completed?

It was already in the newer version

nbAmit, 2022-11-24:
Facing the same issue in version `"react-native-ble-plx": "^2.0.3",`

correadevuk, 2022-12-19:
Expo users: 
Upgraded @config-plugins/react-native-ble-plx to version 5.0.0, and the warning went away.

TheLukaDragar, 2022-12-24:
> 

yea right i dont belive it. doesent work for me

kleydon, 2023-03-09:
I'm encountering this problem as well, on 2.03.