dotintent/react-native-ble-plx #1007: Question

utkarshpandey12, 2022-09-01:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [yes] I am running the latest version
- [yes] I checked the documentation and found no answer
- [yes] I checked to make sure that this issue has not already been filed
- [yes] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

I am able to scan the device and then by filtering the device name I am trying to connect to it and then lists its services and characteristics. using the below  scanAndConnect() method code 

` //
    scanAndConnect() {
    //console.log("entrou", this.manager)
    this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
            // Handle error (scanning will be stopped automatically)
            return
        }
        if (device.name === 'HMSoft' || 
            device.name === 'HMSoft') {
            
            // Stop scanning as it's not necessary if you are scanning for one device.
            this.manager.stopDeviceScan();
            // Proceed with connection.
            
            device.connect()
            .then((device) => {
             return device.discoverAllServicesAndCharacteristics()
            }).then((device) => {
       // Do work on device with services and characteristics
             device.services().then(services => console.log(services))
             //const services = device.services();
             //console.log(services)
            }).catch((error) => {
        // Handle errors
        console.log(error)
            });

            
        }
        // Check if it is a device you are looking for based on advertisement data
        // or other criteria.
        //console.log(device.name)
    });`

My services log is 
[{"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_errorCodesToMessagesMapping": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 4}, "deviceID": "F0:C7:7F:FD:39:AF", "id": 1, "isPrimary": true, "uuid": "00001800-0000-1000-8000-00805f9b34fb"}, {"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_errorCodesToMessagesMapping": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 4}, "deviceID": "F0:C7:7F:FD:39:AF", "id": 7, "isPrimary": true, "uuid": "00001801-0000-1000-8000-00805f9b34fb"}, {"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_errorCodesToMessagesMapping": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 4}, "deviceID": "F0:C7:7F:FD:39:AF", "id": 10, "isPrimary": true, "uuid": "0000180a-0000-1000-8000-00805f9b34fb"}, {"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_errorCodesToMessagesMapping": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 4}, "deviceID": "F0:C7:7F:FD:39:AF", "id": 20, "isPrimary": true, "uuid": "0000ffe0-0000-1000-8000-00805f9b34fb"}]

**Am I headed in the right direction?. Is this the log about the devices services and characteristics? 
If yes how can I write values to a particular characteristics after this step? Please guide with a small code demo. **
@corymsmith 
@below 
@bitcrumb 
@shabith 
@Kaspa

utkarshpandey12, 2022-09-06:
@corymsmith 
@below 
@Kaspa 
@bitcrumb 
anyone?

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska