dotintent/react-native-ble-plx #1008: Expo app crashes immediately if BleManager is used

t3db0t, 2022-09-05:
I created a fresh expo app and installed `react-native-ble-plx` according to [these instructions](https://tedbot.com/How-to-Build-a-Bluetooth-App-with-Expo-React-Native-5127b444ac2448baab4948e36ff97088) and if I use the line

```
const manager = new BleManager();
```
The app crashes immediately upon building without any errors or messages of any kind. If I uncomment that line, it works fine.

Any ideas what I can check or do?

domenbukovac, 2022-09-05:
Are you using Expo Go? If so you need to build custom client since this package requires custom native code. Check out [@config-plugins/react-native-ble-plx](https://www.npmjs.com/package/@config-plugins/react-native-ble-plx)

t3db0t, 2022-09-05:
@domenbukovac Thanks for your reply! Yes, I am using that package and a custom dev client. I've successfully used it in an app before but this crashing behavior is new and I'm not sure how to debug it.

KovD3v, 2022-09-22:
Solved you have to add this 
[image]

closed 2023-09-27 by dominik-czupryna-withintent