dotintent/react-native-ble-plx #1009: Not working in android 13 SDK 33

hirenitp, 2022-09-05:
## Prerequisites

- [X ] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Expected Behavior

Get a list of nearby BLE devices after starting scanning.

## Current Behavior

After execute startDeviceScan() function it gives a result in android 12 SDK 31 but not  in android 13 SDK 33

* Library version: 2.0.3
* Platform: Android.
* React Native ver: 0.69.5

`{
  "name": "BleConnect",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint .",
    "podinstall": "cd ios && pod install && cd ..",
    "postinstall": "npx jetify; yarn run podinstall;",
    "prettier": "prettier --write \"./src/**/*.{js,md,yml}\"",
    "format": "npm run prettier",
    "format-check": "prettier --list-different \"./src/*.{js,md,yml}\"",
    "clean": "watchman watch-del-all && rm -rf package-lock.json && rm -rf node_modules && rm -rf $TMPDIR/metro-* && rm -rf $TMPDIR/haste-map-* && yarn install && say Project has been cleaned successfully",
    "clean-android": "cd android && ./gradlew clean && cd ../ && yarn add jetifier && npx jetify",
    "generate-icons": "./scripts/icon-generator/icon-generator.sh"
  },
  "dependencies": {
    "@react-native-async-storage/async-storage": "^1.17.10",
    "@react-native-community/netinfo": "^9.3.0",
    "@react-native-community/push-notification-ios": "^1.10.1",
    "@react-native-firebase/app": "^15.4.0",
    "@react-native-firebase/messaging": "^15.4.0",
    "@react-navigation/drawer": "^6.4.4",
    "@react-navigation/native": "^6.0.12",
    "@react-navigation/stack": "^6.2.3",
    "@reduxjs/toolkit": "^1.8.5",
    "apisauce": "^2.1.6",
    "buffer": "^6.0.3",
    "formik": "^2.2.9",
    "lodash.get": "^4.4.2",
    "lodash.isempty": "^4.4.0",
    "moment": "^2.29.4",
    "react": "18.0.0",
    "react-native": "0.69.5",
    "react-native-android-location-enabler": "^1.2.2",
    "react-native-ble-plx": "^2.0.3",
    "react-native-device-info": "^10.0.2",
    "react-native-gesture-handler": "^2.6.0",
    "react-native-image-crop-picker": "^0.38.0",
    "react-native-keyboard-aware-scroll-view": "^0.9.5",
    "react-native-permissions": "^3.6.1",
    "react-native-push-notification": "^8.1.1",
    "react-native-reanimated": "^2.10.0",
    "react-native-render-html": "^6.3.4",
    "react-native-safe-area-context": "^4.3.3",
    "react-native-screens": "^3.17.0",
    "react-native-svg": "^13.1.0",
    "react-native-svg-animated-linear-gradient": "^0.4.0",
    "react-query": "^3.39.2",
    "react-redux": "^8.0.2",
    "reactotron-redux": "^3.1.3",
    "redux": "^4.2.0",
    "redux-persist": "^6.0.0",
    "redux-thunk": "^2.4.1",
    "styled-components": "^5.3.5",
    "yup": "^0.32.11"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@babel/runtime": "^7.12.5",
    "@react-native-community/eslint-config": "^2.0.0",
    "babel-jest": "^26.6.3",
    "babel-plugin-module-resolver": "^4.1.0",
    "eslint": "^7.32.0",
    "jest": "^26.6.3",
    "jetifier": "^2.0.0",
    "metro-react-native-babel-preset": "^0.70.3",
    "react-native-svg-transformer": "^1.0.0",
    "react-test-renderer": "18.0.0",
    "reactotron-react-native": "^5.0.2"
  },
  "jest": {
    "preset": "react-native"
  }
}
`

jdmay2, 2022-09-30:
> I ran into the same problem and managed to solve it by :
> 
> * Adding some permissions to my manifest file
> 
> ```
> <uses-permission android:name="android.permission.BLUETOOTH" />
>   <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
>   <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
>   <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
>   <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
>   <uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
> ```
> 
>  * Asking for more permissions when running my app
> 
> ```
> import { PermissionsAndroid } from "react-native"
> 
> const requestBLEPermissions = async () => {
>   const res = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION)
>   await PermissionsAndroid.requestMultiple([ PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN, PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT])
>   console.log(res)
> }
> ```
> 
> I think android permissions have been changed since android 12 : https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

I am having the issue as well and after attempting this, there are still no beacons showing up. Could there be another permission needed, or possibly a different way of scanning for devices?

MBach, 2022-12-01:
> The solution from @plotopt worked for me

Are you sure of this assertion? It does not work for me with Android 13 when I'm starting a scan and when the Bluetooth is disabled. With Android 12, you can request the user to only have the GPS turned on. Once a user has given permission for SCAN and CONNECT, you can start a scan.

However, it seems that there's a bug in most vendor implementation, because if I understand the docs accordingly, even with the flag: `android:usesPermissionFlags="neverForLocation"` I still need the ACCESS_FINE_LOCATION permission. And it's not beacons

KumailAhmedWellwink, 2023-04-13:
Hi  @MBach did you find any solution for Android 13 my startScanDevice gave null

KumailAhmedWellwink, 2023-04-14:
i found this solution before startScanDevice just to make sure your device location and bluetooth On

dominik-czupryna-withintent, 2023-09-29:
Check Android permissions:
Check if you requested following permissions
PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION eg.
```
const locationAccessGranted = await PermissionsAndroid.request(
       PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
       );
```
PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN
PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT

Check if your AndroidManifest.xml (android/app/src/main/AndroidManifest.xml) includes Android 12+ (SDK 31) and Android up to 11 (SDK 30) permissions:
  ```
    <!-- Android >= 12 -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <!-- Android < 12 -->
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    <!-- common -->
```

closed 2023-09-29 by dominik-czupryna-withintent

little-infinity, 2023-10-01:
> I ran into the same problem and managed to solve it by :
>
> * Adding some permissions to my manifest file
>
> ```
> <uses-permission android:name="android.permission.BLUETOOTH" />
>   <uses-permission [...]

### If anyone is facing this issue in 2023 (For lower version of react-native)

I am using,

        compileSdkVersion = 33
        targetSdkVersion = 33
        
--------------------------------------------------
        React-Native - "react-native": "0.63.2"
--------------------------------------------------
     
       Platform Android 12 & 13

The react-native version in my case is quite old, and I cannot afford to update it to the latest version, as it might break a lot of things in the process. However, if time permits, a better approach would be to update your react-native to the latest stable version. If not, the following information might help you:

The permissions BLUETOOTH_CONNECT and BLUETOOTH_SCAN have been newly added in Android 12 and above. If you have recently updated your targetSdk but have not updated your react-native version, it is likely that the suggested solution will not work for you. This is because BLUETOOTH_CONNECT and BLUETOOTH_SCAN constants are not present in older react-native versions. Attempting to request them directly might result in a **Permission not Null error**.

Work-around for this problem is below:

We are going to make changes to the react-native package. Specifically, we'll add these new constants (BLUETOOTH_CONNECT and BLUETOOTH_SCAN) and make sure they're linked to the right permissions in the device's settings. This way, your app can use these permissions correctly, even if you're using an older version of react-native.

Steps: 
1. Install patch-package (#https://www.npmjs.com/package/patch-package)
` npm install patch-package -D `
                or using yarn
`yarn add patch-package`
2. Open node_modules/react-native/Libraries/PermissionAndroid folder & update NativePermissionsAndroid.js & PermissionsAndroid.js file by replicating this commit  --->  #https://github.com/facebook/react-native/commit/eeb8e5829e183f6b5cd5fd327cf6da03a7db0541
3. Add postinstall script to package.json
```
"scripts": {
  "postinstall": "patch-package"
 }
```
4. Run patch-package to create a .patch file
`npx patch-package react-native or yarn patch-package react-native`
This will create .patch file with name react-native+0.63.2.patch - This should committed in your project.
5. Apply your fix:
`npm install or yarn install`

Hopefully, this works!

AlanCasal, 2023-10-03:
> <uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />

Didn't work for me.

BLUETOOTH_PERMISSION_DENIED: android.permission.BLUETOOTH and android.permission.BLUETOOTH_ADMIN are required

AlanCasal, 2023-10-03:
> > I ran into the same problem and managed to solve it by :
> > 
> > * Adding some permissions to my manifest file
> > 
> > ```
> > <uses-permission android:name="android.permission.BLUETOOTH" />
> >   <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
> >   <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
> >   <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
> >   <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
> >   <uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > * Asking for more permissions when running my app
> > 
> > ```
> > import { PermissionsAndroid } from "react-native"
> > 
> > const requestBLEPermissions = async () => {
> >   const res = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION)
> >   await PermissionsAndroid.requestMultiple([ PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN, PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT])
> >   console.log(res)
> > }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I think android permissions have been changed since android 12 : https://developer.android.com/guide/topics/connectivity/bluetooth/permissions
> 
> ### If anyone is facing this issue in 2023 (For lower version of react-native)
> I am using,
> 
> ```
>    compileSdkVersion = 33
>    targetSdkVersion = 33
> 
>    React-Native - "react-native": "0.63.2"
> 
>    Platforms - Android 12 & 13
> ```
> 
> The react-native version in my case is quite old, and I cannot afford to update it to the latest version, as it might break a lot of things in the process. However, if time permits, a better approach would be to update your react-native to the latest stable version. If not, the following information might help you:
> 
> The permissions BLUETOOTH_CONNECT and BLUETOOTH_SCAN have been newly added in Android 12 and above. If you have recently updated your targetSdk but have not updated your react-native version, it is likely that the suggested solution will not work for you. This is because BLUETOOTH_CONNECT and BLUETOOTH_SCAN constants are not present in older react-native versions. Attempting to request them directly might result in a **Permission not Null error**.
> 
> Work-around for this problem is below:
> 
> We are going to make changes to the react-native package. Specifically, we'll add these new constants (BLUETOOTH_CONNECT and BLUETOOTH_SCAN) and make sure they're linked to the right permissions in the device's settings. This way, your app can use these permissions correctly, even if you're using an older version of react-native.
> 
> Steps:
> 
> 1. Install patch-package (#https://www.npmjs.com/package/patch-package)
>    `npm install patch-package -D`
>    or using yarn
>    `yarn add patch-package`
> 2. Open node_modules/react-native/Libraries/PermissionAndroid folder & update NativePermissionsAndroid.js & PermissionsAndroid.js file by replicating this commit  --->  #[facebook/react-native@eeb8e58](https://github.com/facebook/react-native/commit/eeb8e5829e183f6b5cd5fd327cf6da03a7db0541)
> 3. Add postinstall script to package.json
> 
> ```
> "scripts": {
>   "postinstall": "patch-package"
>  }
> ```
> 
> 4. Run patch-package to create a .patch file
>    `npx patch-package react-native or yarn patch-package react-native`
>    This will create .patch file with name react-native+0.63.2.patch - This should be committed in your project.
> 5. Apply your fix:
>    `npm install or yarn install`
> 
> Hopefully, this works!

Didn't work for me.

BLUETOOTH_PERMISSION_DENIED: android.permission.BLUETOOTH and android.permission.BLUETOOTH_ADMIN are required

JayarajDeveloper, 2023-10-17:
Hi i am using 
 compileSdkVersion = 33
   targetSdkVersion = 33

   Platforms - Android 12 & 13
   
   Ionic - facing crashing issue while startDeviceScan. please help me