dotintent/react-native-ble-plx #1011: iOS. Device is connected but then is disconnected in 3 - 5 seconds

olegderecha, 2022-09-07:
## Expected Behavior

The device is connected and is not disconnected automatically.

## Current Behavior

The device is connected but then is disconnected in 3 - 5 seconds

### Steps to Reproduce

1. Created a clean react native project.
2. Scanned for online devices.
3. Connected to the device.
4. After several seconds the device is not connected anymore.
5. If reconnect again, the picture is the same - it is connected for several seconds and then is disconnected automatically. The disconnection event does not contain any error (event is fired, but the error is _null_)

Tried on some other demo projects and used other devices (tested by customers) - the behavior is the same.
The issue is **not** present on Android devices.

### Context

* Library version: 2.0.3
* Platform: iOS 16.
* Device: iPhone 12 Pro
* BLE device: Fitness tracker Samsung Gear Fit2 (7987)

tomaseliasm, 2022-09-22:
i'm having the same issue, have you found any workaround @olegderecha ?

tomaseliasm, 2022-09-23:
in my case we made changes to the FW of the ble device in order to fix the disconnect issue.
https://devzone.nordicsemi.com/f/nordic-q-a/92003/getting-continuous-disconnects-with-ios-16/386457

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska