dotintent/react-native-ble-plx #1014: Android startDeviceScan does not discover all BLE devices

dominikwagner, 2022-09-15:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [X] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question
Hi

we are using this library to scan for Nuki smartlocks and this is working great for ios and Android (Google Pixel 4, Android 13).
On ios we are also finding all other nearby BLE devices. On Android there are BLE devices which are not found with `startDeviceScan`.

When I start a scan with nrf connect on Android all BLE devices are found.
Do you have any idea why some devices are missing? 

Thanks!

mathias5r, 2022-09-21:
It happens really often with me too. My device is just found within `startDeviceScan` after the third or fouth tries, even though nRF connect find it instantly.

dominikwagner, 2022-09-21:
@mathias5r  are you scanning with scan mode `LOW_LATENCY`? 
The scan mode controls how long and with which frequency your device is searching for BLE devices. The default mode is `LOW_ENERGY`. In the low energy mode Android scans for a very short time (I believe 0.5 seconds) and then pauses for a long time (I believe 4 seconds). Depending on how often the BLE device is advertising it may take a long time on Android to find the BLE device. When you change the scan mode to low latency your device is scanning without interruption and so the BLE device is found faster (but it consumes also way more energy).

But we had another hidden problem:
Android 13 introduced [neverForLocation](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions) permission flag. If you set this flag on the `BLUETOOTH_SCAN`permission you can remove the `ACCESS_FINE_LOCATION` permission for ble scanning (if locaiton is not otherwise needed). In the docs you can find the following note `If you include neverForLocation in your android:usesPermissionFlags, some BLE beacons are filtered from the scan result`.

We never thought about that because we have not inserted this flag in our manifestfile. But we used a library which included this flag (we found it looking at the merged manifest file....). After removing this flag with `tools:remove` on the bluetooth scan permission we found the BLE device which was missing in the Android scan results:
`<uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:remove="android:usesPermissionFlags" />`

So if you can find not all BLE devices on Android 13 and you are sure that the missing device is a BLE device than check if ´neverForLocation` is not in use. As written in the docs this could result in missing devices in the scan result.

closed 2022-11-11 by dominikwagner

pariharkaran, 2024-04-24:
Facing same issue on Google Pixel 7 (Android 14), above solution didn't worked for me. If anyone found solution please comment here

jkaos92, 2024-05-07:
@mathias5r amazing, thank you, it worked even in another bluetooth library which wasn't working on Android 14 but worked on Android 12. Same reason probably. 

```
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:remove="android:usesPermissionFlags" />
```

@pariharkaran not sure what you tried, but tools:remove="android:usesPermissionFlags" did the trick for us.