dotintent/react-native-ble-plx #1015: [IOS] app freezes when writing characteristic data

Jeremika, 2022-09-15:
## Expected Behavior

The app correctly writes and returns a response after I call writeWithoutResponse on ios.

## Current Behavior
The ios app freezes after i call writeWithoutResponse. In my code i call this function two times. The first time the app returns the characteristic but i don't get the value i wrote on the characteristic. The second time i try to write to the device, the program freezes and it never returns a response. 

The program works fine on android and data is returned each time i call writeWithoutResponse.

### Steps 

1. connect the manager to the device
2. called discoverAllServicesAndCharacteristics 
3. call writeWithoutResponse to ota characteristic to get access to write on the device 
4. call writeWithoutResponse to change the value of a characteristic

### Context

* Library version: 2.0.3
* Platform: iOS
* Version: 12.5.5
*  Device: Iphone 6

My package json:

```json
{
  "name": "efuzeapp",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint . --ext .js,.jsx,.ts,.tsx"
  },
  "dependencies": {
    "@miblanchard/react-native-slider": "^2.1.0",
    "@react-native-async-storage/async-storage": "^1.17.9",
    "@react-navigation/bottom-tabs": "^6.3.2",
    "@react-navigation/native": "^6.0.11",
    "@react-navigation/native-stack": "^6.7.0",
    "axios": "^0.27.2",
    "base-64": "^1.0.0",
    "buffer": "^6.0.3",
    "convert-string": "^0.1.0",
    "lodash.debounce": "^4.0.8",
    "mobx": "^6.6.1",
    "mobx-react-lite": "^3.4.0",
    "mobx-state-tree": "^5.1.6",
    "moment": "^2.29.4",
    "react": "18.0.0",
    "react-native": "0.69.4",
    "react-native-ble-plx": "^2.0.3",
    "react-native-chart-kit": "^6.12.0",
    "react-native-customisable-alert": "^0.1.20",
    "react-native-paper": "^5.0.0-rc.4",
    "react-native-safe-area-context": "^4.3.1",
    "react-native-screens": "^3.15.0",
    "react-native-svg": "^13.0.0",
    "react-native-vector-icons": "^9.2.0"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@babel/runtime": "^7.12.5",
    "@react-native-community/eslint-config": "^2.0.0",
    "@tsconfig/react-native": "^2.0.2",
    "@types/convert-string": "^0.1.1",
    "@types/jest": "^26.0.23",
    "@types/lodash.debounce": "^4.0.7",
    "@types/react-native": "^0.69.3",
    "@types/react-native-vector-icons": "^6.4.11",
    "@types/react-test-renderer": "^18.0.0",
    "@typescript-eslint/eslint-plugin": "^5.29.0",
    "@typescript-eslint/parser": "^5.29.0",
    "babel-jest": "^26.6.3",
    "eslint": "^7.32.0",
    "jest": "^26.6.3",
    "metro-react-native-babel-preset": "^0.70.3",
    "react-test-renderer": "18.0.0",
    "typescript": "^4.4.4"
  },
  "jest": {
    "preset": "react-native",
    "moduleFileExtensions": [
      "ts",
      "tsx",
      "js",
      "jsx",
      "json",
      "node"
    ]
  }
}

```
* My Code

```js
const writeCharacteristic = () => {
    //get the ota key
    setResettingSensor(true);
    if (!deviceToReset) {
      console.error('The device to reset is not defined');
      setResettingSensor(false);
      return;
    }
    const device = deviceToReset;
    const otaKeyHex = getOtaKeyValue(device);
    let otaKeyInt = parseInt(otaKeyHex, 16).toString();
    let msb = (otaKeyInt >> 8) & 0xff;
    let lsb = otaKeyInt & 0xff;
    const otaKey_chaHex = [lsb, msb];
    const otaKey_cha = Buffer(otaKey_chaHex).toString('base64'); // value of the ota key used to write on the device
    let resetCh = '1';
    msb = (resetCh >> 8) & 0xff;
    lsb = resetCh & 0xff;
    const resetChHex = [lsb, msb];
    const reset_cha = Buffer(resetChHex).toString('base64'); // reset fuse value used to write the characteristic value
    device
      .connect()
      .then(device => {
        console.log('Connected to the device');
        // discover charactetirstics
        return device.discoverAllServicesAndCharacteristics();
      })
      .then(async device => {
        try {
          /*
            Here i write to the ota characteristic
            on android i recieve the characteristic in responseOta.
            on ios it does the same but the characteristic.value is null
          */
          const responseOta =
            await device.writeCharacteristicWithoutResponseForService(
              SERVICE_UUID,
              OTA_CHARACTERISTIC_UUID, //ota key characteristic uuid
              otaKey_cha,
            );
          console.log('OTA RESPONSE');
          console.log(responseOta);
          /*
              Here is where i write to the characteristic that i want to reset.
              On android it works fine but on ios it freezes and never returns a value
          */
          const responseFuse =
            await device.writeCharacteristicWithoutResponseForService(
              SERVICE_UUID,
              RESET_FUSE_CHARACTERISTIC_UUID, // reset fuse characteristic uuid
              reset_cha,
            );
          console.log('RESET FUSE RESPONSE');
          console.log(responseFuse);
          await device.cancelConnection();
          setDeviceToReset(undefined);
          showAlert({
            title: 'Sensor successfully reset',
            alertType: 'success',
          });
        } catch (error) {
          console.log('Error while writting to the sensor');
          console.error(error);
          await device.cancelConnection();
        } finally {
          setResettingSensor(false);
        }
      })
      .catch(error => {
        console.log('error while connecting to the sensor');
        console.error(error);
        setResettingSensor(false);
      });
  };
```

anyky, 2022-11-14:
Is there any solution to this?

VetInstant, 2023-06-03:
writeCharacteristicWithoutResponseForService to writeCharacteristicWithResponseForService try this one

dominik-czupryna-withintent, 2023-09-29:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs. Please confirm if the issue still persists and create a new issue if it still does.