dotintent/react-native-ble-plx #1019: Why doesn't monitor notify for every value change on the characteristic?

jurck222, 2022-09-24:
I am trying to make an app that receives data from a device but when i run the app and send the commands the monitor sometimes outputs the changed value and other times when i run the app it just doesnt do anything.

this is my code for connecting.

`function scanAndConnect() {
    BLTManager.startDeviceScan(null, null, (error, device) => {
        if (error) {
            return
        }
        if (device.name=='Audio PCM Streamer') {
            console.log(device.name);
            BLTManager.stopDeviceScan();
            device.connect()
            .then(async (device) => {
              device.requestMTU(251);
              setConnectedDevice(device);
              return device.discoverAllServicesAndCharacteristics()
            })
            .then( async (device) => {
              const services = await device.services();
              const characteristics = await services[services.length-1].characteristics();
              console.log("Characteristics:",characteristics.length);
              characteristics[0].monitor((err, update) => {
                if (err) {
                  console.log(`characteristic error: ${err}`);
                  console.log(JSON.stringify(err));
                } else {
                  console.log("Is Characteristics Readable:",update.isReadable);
                  console.log("Received data" ,update.value);
                }
              });
              await setCaptureActionAUDIO(1,1600*4,device);
              readDataAUDIO(device);
              return device;
            })
           }
        });
}`

and these are functions that i use:
`async function setCaptureActionAUDIO(action, size,device){
    let n = 0;
    let pktsize  = 9;
    let pktdata = new Uint8Array(pktsize);
    for(let i = 0; i<9;i++){
      pktdata[i]= 0x00;
    }
    
    // generate packet
	  pktdata[0] = 0xA5;
	  pktdata[1] = 0x52;
	  pktdata[2] = 0x05;
	  pktdata[3] = action;
    
    n = 4;
    let tmp = size;
    
    pktdata[n+0] = parseInt(tmp/2**24);
	  tmp = tmp - pktdata[n+0]*(2**24)
	  pktdata[n+1] = parseInt(tmp/2**16);
	  tmp = tmp - pktdata[n+1]*(2**16)
	  pktdata[n+2] = parseInt(tmp/2**8);
	  tmp = tmp - pktdata[n+2]*(2**8)
	  pktdata[n+3] = parseInt(tmp);
    txPktBLE(pktdata, pktsize, 0x0012,device);
    console.log("-> BLE: setting action mode"+action+"S \n"+size);
    
  }`

`async function txPktBLE(pkt, pktlen, hnd,device){
    pkt[pktlen-1]=0x00;
    let crc0 = 0;
    for(let i = 0; i<pktlen-1; i++){
      if((crc0  + pkt[i]) > 255){
        crc0 = crc0 + pkt[i] - 256;
      }
      else{
        crc0 = crc0 + pkt[i];
      }
    }
    pkt[pktlen-1] = (crc0^0xFF);
    if((pkt[pktlen-1] + 1) > 255){
      pkt[pktlen-1] = 0;
    }
    else{
      pkt[pktlen-1] = pkt[pktlen-1] + 1;
    }
    g_ble_rx_rflag = 0
    var b64encoded = btoa(String.fromCharCode.apply(null, pkt));
    try{
      await device.writeCharacteristicWithoutResponseForService(SERVICE_UUID,CHARACTERISTIC_UUID,b64encoded);
    }catch(exceptionVar){
      console.log(exceptionVar);
    }
}`

`async function readDataAUDIO(device){
    let n = 0;
    let pktsize = 9;
    let pktdata = new Uint8Array(pktsize);
    let bdata = new Uint8Array(2*512*1024);
    let fdata = [];
    let pdata = [];
    for(let i = 0; i<512*1024; i++){
      fdata.push(0);
      pdata.push(0);
    }
    for(let i = 0;i<pktsize;i++){
      pktdata[i] = 0x00;
    }
    pktdata[0] = 0xA5;
    pktdata[1] = 0x53;
    pktdata[2] = 0x00;

    await txPktBLE(pktdata, 4, 0x0012,device);
    
    console.log("-> BLE: waiting for reply...")
    g_ble_rx_rflag=1;
    if(wait4RespBLE(2000) == 0){
      console.log("-> BLE: write timeout")
      return;
    }
    console.log("no timeout");
    
    n = 3;	
    let cap_size = 0;
    let cbyte = (g_ble_rx_pkt[n + 0]);
    cap_size = cap_size + parseInt(cbyte)*(2**24);
    cbyte = (g_ble_rx_pkt[n + 1]);
    cap_size = cap_size + parseInt(cbyte)*(2**16);
    cbyte = (g_ble_rx_pkt[n + 2]);
    cap_size = cap_size + parseInt(cbyte)*(2**8);
    cbyte = (g_ble_rx_pkt[n + 3]);
    cap_size = cap_size + parseInt(cbyte);
    
    console.log("-> BLE: preparing for download of "+cap_size+" B");

    let ridx = 0;
   
      pktdata[0] = 0xA5;
      pktdata[1] = 0x55;
      pktdata[2] = 0x04;
      n = 3;
      let tmp = ridx;
      
      pktdata[n + 0] = parseInt(tmp/2**24)
      tmp = tmp - pktdata[n + 0]*(2**24)
      pktdata[n + 1] = parseInt(tmp/2**16)
      tmp = tmp - pktdata[n + 1]*(2**16)
      pktdata[n + 2] = parseInt(tmp/2**8)
      tmp = tmp - pktdata[n + 2]*(2**8)
      pktdata[n + 3] = parseInt(tmp);
      
      
      await txPktBLE(pktdata, 8, 0x0012,device);
}`

AzranAzwer, 2023-02-16:
did you find any ?

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska