dotintent/react-native-ble-plx #1022: I'm not getting my characteristic to be able to read the device data and show it on the screen

MuriloBarranco, 2022-10-14:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

Please describe the behavior you are expecting

## Current Behavior

What is the current behavior?

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

I'm not getting my characteristic to be able to read the device data and show it on the screen
I'm getting the following return.

{
   "index":0,
   "item":{
      "_manager":{
         "_activePromises":[
            "Object"
         ],
         "_activeSubscriptions":[
            "Object"
         ],
         "_errorCodesToMessagesMapping":[
            "Object"
         ],
         "_eventEmitter":[
            "NativeEventEmitter"
         ],
         "_scanEventSubscription":[
            "EmitterSubscription"
         ],
         "_uniqueId":0
      },
      "id":"80:30:DC:0D:01:40",
      "isConnectable":null,
      "localName":"HMSoft",
      "manufacturerData":"SE2AMNwNAUA=",
      "mtu":23,
      "name":"HMSoft",
      "overflowServiceUUIDs":null,
      "rssi":-65,
      "serviceData":null,
      "serviceUUIDs":[
         "0000ffe0-0000-1000-8000-00805f9b34fb"
      ],
      "solicitedServiceUUIDs":null,
      "txPowerLevel":0
   },
   "separators":{
      "highlight":[
         "Function highlight"
      ],
      "unhighlight":[
         "Function unhighlight"
      ],
      "updateProps":[
         "Function updateProps"
      ]
   }
}

I can only get the serviceUUIDs but I still need to get the CHARACTERISTIC and the UUID

MuriloBarranco, 2022-10-14:
I get the following error [BleError: Service 228FE733-2BAB-CE6F-9772-FE0DD93D81A1 for device ? not found]

closed 2022-10-25 by MuriloBarranco

Zanetti85, 2022-10-29:
how did you resolve?