dotintent/react-native-ble-plx #1029: I can connect but cannot write on the device

AzranAzwer, 2022-11-02:
## i can connect to the device but cannot write.

## after I connect to the device i'm getting the response. But after that when try to send data, iis giving me the attached error.

my code is below.

        const conectedDevice = await manager.connectToDevice(device.id)

        await device.discoverAllServicesAndCharacteristics()

        const services = await device.services()

        const characteristics = await services[1].characteristics()

        const characteristic = _.find(
          characteristics,
          ({isWritableWithResponse}) => isWritableWithResponse === true,
        )

       const base64Data = base64.encode(
          '{ "Command": "SALE", "Amount": "000000001000" }',
        )

        const response = await characteristic.writeWithResponse(base64Data)

        // const response =await device.writeCharacteristicWithResponseForService(
        //   '"4DF91029-B356-463E-9F48-BAB077BF3EF5"',
        //   characteristic.uuid,
        //   base64Data,
        // )

        // const response = await manager.writeCharacteristicWithResponseForDevice(
        //   characteristic.deviceID,
        //   characteristic.serviceUUID,
        //   characteristic.uuid,
        //   base64Data,
        // )
        console.log('response', response)

[image]

ziedhajsalah, 2022-11-28:
@AzranAzwer , did you manage to fix this issue?

AzranAzwer, 2022-11-28:
no

AzranAzwer, 2023-01-17:
Hi All I have fixed this issue

Actually my encryption was wrong after I tried for couple of methods, I got the exact encryption way.

closed 2023-01-17 by AzranAzwer

hoanghai9650, 2023-04-08:
can you share how did you encrypt ?