dotintent/react-native-ble-plx #1035: Package is not working on Android 12

asifrashidny, 2022-11-10:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version - Yes
- [ ] I checked the documentation and found no answer - Yes
- [ ] I checked to make sure that this issue has not already been filed - Yes

## Expected Behavior

The package should work on android 12. 

## Current Behavior

I had to upgrade to Android 12 but this package is not working on it. 

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. ...

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://dotintent.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: X.Y.Z
* Platform: Android/iOS.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

```js
function example(): string {
  return "Your code sample"
}
```

sdandois, 2022-11-18:
Duplicate of #1009.

dominik-czupryna-withintent, 2023-09-29:
Check Android permissions:
Check if you requested following permissions
PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION eg.
```
const locationAccessGranted = await PermissionsAndroid.request(
       PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
       );
```
PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN
PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT

Check if your AndroidManifest.xml (android/app/src/main/AndroidManifest.xml) includes Android 12+ (SDK 31) and Android up to 11 (SDK 30) permissions:
    
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

closed 2023-09-29 by dominik-czupryna-withintent