dotintent/react-native-ble-plx #1040: TypeError: Cannot read property 'startDeviceScan' of null

omerkars78, 2022-12-23:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Please describe the question.

omerkars78, 2022-12-23:
I need to get the detailed bluetooth data of the surrounding devices when I press the button. I use the react-native-ble-plx library for this, but this library does not integrate with android 12. I made some necessary permisson adjustments for this. On manifest.xml file and app.js files. When I run the code, I get the following error about the "scanDeviceStart" method.

`TypeError: Cannot read property 'startDeviceScan' of null`

What would you suggest me to solve this error? my codes are below 

```
import React, { useState, useEffect } from 'react';
    import {
    SafeAreaView,
    View,
    Text,
    Button,
    StyleSheet,
    PermissionsAndroid ,
    FlatList,
    } from 'react-native';
    import BleManager from 'react-native-ble-plx';
    import { PERMISSIONS, requestMultiple } from 'react-native-permissions';
    import DeviceInfo from 'react-native-device-info';
    
    const requestPermissions = async () => {
    if (Platform.OS === 'android') {
    const apiLevel = await DeviceInfo.getApiLevel();
    
    if (apiLevel < 31) {
      const granted = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
        {
          title: 'Location Permission',
          message: 'Bluetooth Low Energy requires Location',
          buttonNeutral: 'Ask Later',
          buttonNegative: 'Cancel',
          buttonPositive: 'OK',
        },
      );
    } else {
      const result = await requestMultiple([
        PERMISSIONS.ANDROID.BLUETOOTH_SCAN,
        PERMISSIONS.ANDROID.BLUETOOTH_CONNECT,
        PERMISSIONS.ANDROID.ACCESS_FINE_LOCATION,
      ]);
    
      const isGranted =
        result['android.permission.BLUETOOTH_CONNECT'] ===
        PermissionsAndroid.RESULTS.GRANTED &&
        result['android.permission.BLUETOOTH_SCAN'] ===
        PermissionsAndroid.RESULTS.GRANTED &&
        result['android.permission.ACCESS_FINE_LOCATION'] ===
        PermissionsAndroid.RESULTS.GRANTED;
    }
    
    }
    };
    
    const App = () => {
    const [devices, setDevices] = useState([]);
    const [scanning, setScanning] = useState(false);
    const [bleManager, setBleManager] = useState(null);
    
    function requestBluetoothPermissions(cb) {
    if (Platform.OS === 'android') {
    DeviceInfo.getApiLevel().then((apiLevel) => {
    if (apiLevel < 31) {
    PermissionsAndroid.request(
    PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
    {
    title: 'Location Permission',
    message: 'Bluetooth Low Energy requires Location',
    buttonNeutral: 'Ask Later',
    buttonNegative: 'Cancel',
    buttonPositive: 'OK',
    },
    ).then((granted) => {
    cb(granted === PermissionsAndroid.RESULTS.GRANTED);
    });
    } else {
    requestMultiple([
    PERMISSIONS.ANDROID.BLUETOOTH_SCAN,
    PERMISSIONS.ANDROID.BLUETOOTH_CONNECT,
    PERMISSIONS.ANDROID.ACCESS_FINE_LOCATION,
    ]).then((result) => {
    const isGranted =
    result['android.permission.BLUETOOTH_CONNECT'] === PermissionsAndroid.RESULTS.GRANTED &&
    result['android.permission.BLUETOOTH_SCAN'] === PermissionsAndroid.RESULTS.GRANTED &&
    result['android.permission.ACCESS_FINE_LOCATION'] === PermissionsAndroid.RESULTS.GRANTED;
    cb(isGranted);
    });
    }
    });
    } else {
    cb(true);
    }
    }
    
    useEffect(() => {
        if (!bleManager) {
        requestBluetoothPermissions((isGranted) => {
        if (isGranted) {
        const manager = new BleManager();
        setBleManager(manager);
        manager.startDeviceScan(  [serialUUIDs.serviceUUID],
          {scanMode: ScanMode.LowLatency}, (error, device) => {
        if (error) {
        console.log(error.message);
        return;
        }
    
        console.log(device.name);
        console.log(device.id);
    
        if (device.name) {
          setDevices((prevDevices) => [...prevDevices, device]);
        }
        });
        setScanning(true);
        }
        });
        }
    }, []);
    
    return (
    <SafeAreaView>
      <View>
          {scanning ? (
          <Button
          title="Stop Scanning"
          onPress={() => {
          bleManager.stopDeviceScan();
          setScanning(false);
          }}
          />
          ) : (
          <Button
          title="Start Scanning"
          onPress={() => {
          requestBluetoothPermissions((isGranted) => {
          if (isGranted) {
          setDevices([]);
          bleManager.startDeviceScan([serialUUIDs.serviceUUID],
            {scanMode: ScanMode.LowLatency},(error, device) => {
          if (error) {
          console.log(error.message);
          return;
          }
          console.log(device.name);
          console.log(device.id);
    
          if (device.name) {
            setDevices((prevDevices) => [...prevDevices, device]);
          }
          });
          setScanning(true);
          }
          });
          }}
          />
          )}
      </View>
        <FlatList
        data={devices}
        renderItem={({ item }) => (
        <Text>
        {item.name} ({item.id})
        </Text>
        )}
        keyExtractor={(item) => item.id}
        />
    </SafeAreaView>
    );
    };
    
    export default App;
```

omerkars78, 2022-12-23:
What would you suggest me to solve this error? my codes are below

yosukapro, 2023-02-15:
Same problem here. Did you resolve it yet @omerkars78 ?

dominik-czupryna-withintent, 2023-09-29:
```
import {BleManager, ScanMode} from 'react-native-ble-plx';
```
instead
```
import BleManager from 'react-native-ble-plx';
```
and i think
``` 
const {current: bleManager} = useRef(new BleManager());
```
fits better for yours case than
```
const [bleManager, setBleManager] = useState(null);
```

closed 2023-09-29 by dominik-czupryna-withintent