dotintent/react-native-ble-plx #1041: monitorCharacteristicForDevice divides the response into 2

apoorv-1310, 2022-12-25:
## Prerequisites

- [y] I am running the latest version - yes
- [y] I checked the documentation and found no answer - yes, there is no such response
- [y] I checked to make sure that this issue has not already been filed - exact same issue is not found anywhere

## Expected Behavior

it should return whole data at once 
**Abhishek,2,10,0,1,40,91,25**

## Current Behavior

it is dividing the response into 2 and returning in 2 subscriptions
**response 1 - Abhishek,2,10,0,1,40**
**response 2 - 91,25**

```js
  React.useEffect(() => {
    const startListener=async () => {
      if(selectedDevice?.connected) {
        console.info('CONNECTED');
        const services=await selectedDevice.device.discoverAllServicesAndCharacteristics();
        const characteristic=await getServicesAndCharacteristics(services);
        bleManager.monitorCharacteristicForDevice(selectedDevice.device.id,SERVICEID,CHARACTERSTICID,(error: BleError|null,characteristic: Characteristic|null) => {
          console.log("ERROR",error);
          console.log("characteristic",characteristic!.value)
        })
        // bleManager.monitorCharacteristicForDevice(selectedDevice.device.id,SERVICEID,CHARACTERSTICID).then((value:Characteristic)=>{
        //   console.log("value",value)
        // })
      }
    }
    startListener()
  },[selectedDevice])
```

apoorv-1310, 2022-12-27:
@Cierpliwy can you help me out ?

dominik-czupryna-withintent, 2023-09-29:
Could you say something more? monitorCharacteristicForDevice returns characteristics one at a time. Have you tried increasing the MTU value? You are talking about an Android/iOS device?

closed 2023-11-02 by dominik-czupryna-withintent