dotintent/react-native-ble-plx #1042: How can i get ibeacon's (major and minor) datas.

omerkars78, 2022-12-29:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Please describe the question.

omerkars78, 2022-12-29:
I can get only id,name and rssi but i need to get major and minor values. How can i get major and minor datas. Btw my code is 
```
import React, { useState } from 'react';
import {
  Button,
  SafeAreaView,
  ScrollView,
  StatusBar,
  StyleSheet,
  Text,
  useColorScheme,
  View,
} from 'react-native';

import {
  Colors,
  DebugInstructions,
  Header,
  LearnMoreLinks,
  ReloadInstructions,
} from 'react-native/Libraries/NewAppScreen';

import {BleManager} from 'react-native-ble-plx';
import useBLE from './useBLE';
var Buffer = require('buffer/').Buffer;
import base64 from 'react-native-base64';
const Parser = require("binary-parser").Parser;

const App = () => {
  const manager = new BleManager();
  const {requestPermissions} = useBLE();

  useState(() => {
    requestPermissions((isGranted) => {
      if (!isgranted) {
        alert("android permission granted: " + isGranted);
      }
    });
  }, []);

  // const ibeacon_format = Struct(
  //   "uuid" / Array(16, Byte),
  //   "major" / Int16ub,
  //   "minor" / Int16ub,
  //   "power" / Int8sl,
  // )

  const ibeaconFormat = new Parser()
    .int16("major")
    .int16("minor");

  function deneme() {
    manager.startDeviceScan(null, null, (error, device) => {
      
      console.log("device id: ",device.id);
      console.log("device name: ", device.localName);
      console.log("device rssi: ", device.rssi);
      // console.log(device.readCharacteristicForService());
      console.log("--------------------");

      // let data = new Buffer(device.manufacturerData.toString("base64"));
      // console.log(data);
      // let data = device.manufacturerData[0x004C];
      // let beacon = ibeacon_format.parse(data);
      // console.log(beacon);

      // let data = Buffer.from(device.manufacturerData, "base64");
      // console.log(ibeaconFormat.parse(data));

    

      if (error) {
        console.log(error);
      }
    });

  }
  
  function durdur() {
    manager.stopDeviceScan();
  }

  return (
    <View>
      <Text>Hello World</Text>
      <Button title="Deneme" onPress={deneme} />
      <Button title="Durdur" onPress={durdur} />
    </View>
  );
};

export default App;
```

np-singh, 2023-02-01:
Hey @omerkars78 , Were you able to solve this?

omerkars78, 2023-02-01:
yes @np-singh  this is code is solved this problem. 

```
import {BleManager} from 'react-native-ble-plx';
import RNFetchBlob from 'rn-fetch-blob';
import {stringToBytes} from 'convert-string';
 const base64 = RNFetchBlob.base64;
            const advertisingData = stringToBytes(
              base64.decode(device.manufacturerData),
            );

            const rssi = device.rssi;
            const name = device.localName;
            const major = advertisingData[21]; // => this is major data
            const minor = advertisingData[23]; // this is minor data 
```

closed 2023-09-26 by dominik-czupryna-withintent