dotintent/react-native-ble-plx #1047: duplicates sneak there way into the response, even when allowDuplicates flag is set to false.

Vercjames, 2023-01-12:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x ] I am running the latest version
- [x ] I checked the documentation and found no answer
- [ x] I checked to make sure that this issue has not already been filed

## Expected Behavior

When scanning, duplicates should not appear during the `manager.startDeviceScan` event provided that the `{ allowDuplicates: false }` flag is included. 

## Current Behavior

Every so often a duplicate does appear

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. create a blx application that scans for bluetooth devices. 
2. use another device to broadcast an bluetooth single. 
3. log the responses from the manager.startDeviceScan.
4. toggle blx application until it messes up.

harrisol22, 2023-01-26:
This is happening to me as well except it isn't occasional - I'm currently looking at the same device scanned 216 times even though { allowDuplicates: false }.

dominik-czupryna-withintent, 2023-09-29:
Currently this is an iOS only feature https://dotintent.github.io/react-native-ble-plx/#scanoptions

closed 2023-09-29 by dominik-czupryna-withintent