dotintent/react-native-ble-plx #1048: Bundle Release Mode causes device discovery and connection to stop

spit4520, 2023-01-22:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [X ] I am running the latest version
- [ X] I checked the documentation and found no answer
- [ X] I checked to make sure that this issue has not already been filed

## Expected Behavior

In  bundle release mode, I expect the app to behave identically to the dev mode

## Current Behavior

What is the current behavior?

When I bundle release all of the BLE functionality stops working, I have added in all proguard rules, all permissions, manually enabled all permissions. I have my code running in two separate version one on RN: 71 and one on RN: 62 in two separate projects and neither of them work

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

npx react-native run-android --mode=release (RN 71)
npx react-native run-android --variant=release (RN 62)

I have no idea how I got here so I don't have better instructions

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://dotintent.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 2.0.3
* Platform: Android
There really isn't anything else to show, the device never does anything

This issue is blocking production and I am willing to pay for support, please email me: scott (a.t.) spitler (dota . . ) dev

spit4520, 2023-01-22:
This [issue](https://github.com/dotintent/react-native-ble-plx/issues/1009) did not solve it for me, I followed all of the suggested steps for Android 12, issue still persists

dominik-czupryna-withintent, 2023-09-29:
The latest version (3.0.0) of the library has been tested in the release version and no error was found. Please confirm if the issue still persists and create a new issue if it still does.

closed 2023-09-29 by dominik-czupryna-withintent