dotintent/react-native-ble-plx #1049: Bug Report

VirtuozTM, 2023-01-23:
Hello everyone,

I'm contacting you about a problem I can't seem to solve (about, in my opinion, android permissions). 

The problem is the following: I have a mobile application developed in React Native in TypeScript language. Let's take the case, where the problem occurs. I am on a screen and I click on a button. This button makes me navigate to another screen and when I open this screen, I ask for different permissions (such as location, bluetooth etc). I did all my tests on my phone which has an android version 10. Everything works perfectly! However, as soon as I use a friend's phone which has an android 12 version, I get an error when I navigate to the screen. The navigation is fine but when asking for permissions either :
- the application crashes
- the requests are done but the application is totally buggy and I have to restart it.
 
In conclusion, when the application is initialized, from the first installation, the permission requests break the application.

I asked help Dan (from https://www.youtube.com/@DanRNLab) because he did a lot of video on BLE. He recommended that I follow his video to fix the android 12 problems but I had already done that :( and it didn't solve my problem.

So you are my last chance to find the error.

A video showing the process and photos are attached.  

Hoping that you can help me.

[image]

https://user-images.githubusercontent.com/105630116/213995590-ea5d2b90-7478-443b-8f1d-6d6d57e9e445.mp4

[image]
[image]

will-thrive, 2023-01-23:
Hi VirtuozTM.  I have BLE working on Android 13 (and 12 too I believe.). I have checked my android/app/src/main/AndroidManifest.xml file and my BLE code.  The only real difference I can see to yours is that I do NOT have or use BLUETOOTH_ADVERTISE in either.  Are you sure you need this?  I mean, do you need to send advertisements?  Usually advertisments come from the device, i.e. from the server.

VirtuozTM, 2023-01-23:
I remove BLUETOOTH_ADVERTISE but it doesn't change anything :(. I have the impression that BLUETOOTH_CONNECT is ghosted. I have those versions : 
 "react-native-base64": "^0.2.1",
 "react-native-ble-manager": "^8.4.1",
 "react-native-ble-plx": "^2.0.3",
 "react-native-bluetooth-state-manager": "^1.3.3"

VirtuozTM, 2023-01-23:
Btw, when I'm going in the settings of my application on the phone. "Authorizations" are disabled and I can active them manually (as you can see on the screenshot just under)
[image]

will-thrive, 2023-01-23:
You have both "react-native-ble-manager": "^8.4.1" and "react-native-ble-plx": "^2.0.3".  I use only "react-native-ble-plx": "^2.0.3".  You only need one BLE liberary or the other, not both.  Are you actually using react-native-ble-manager?

VirtuozTM, 2023-01-23:
I use only "react-native-ble-plx" as well. I tried "react-native-ble-manager" ealier to compare both librairies. Globally I use : "react-native-base64": "^0.2.1",
"react-native-ble-plx": "^2.0.3",
"react-native-bluetooth-state-manager": "^1.3.3",
"react-native-permissions": "^3.6.1",

I update my version of react-native-permissions from 3.3.1 to 3.6.1 but it doesn't change anything

VirtuozTM, 2023-01-23:
I can add other informations. When I navigate to the screen I use in a 
` useEffect(() => {
    initBleScanner(serviceUUID);
  }, [initBleScanner, serviceUUID]);`

I call an function which name is `initBleScanner`. It looks like : 

`const initBleScanner = useCallback(
    (UUID_DEVICE: string) => {
      requestPermissions((permissionsGranted: boolean) => {
        if (permissionsGranted) {
          // Code pour continuer avec le scan
          const subscription = manager.current.onStateChange(state => {
            if (state === 'PoweredOn') {
              scanDevices(UUID_DEVICE);
              subscription.remove();
            }
          }, true);
        } else {
          // Code pour gérer l'absence de permission
        }
      });
    },
    [scanDevices],
  );
`

Finally, the `requestPermissions((permissionsGranted: boolean)` is : 

`const requestPermissions = async (callback: PermissionCallBack) => {
    if (Platform.OS === 'android') {
      const apiLevel = await DeviceInfo.getApiLevel();
      if (apiLevel < 31) {
        const grantedStatus = await PermissionsAndroid.request(
          PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
        );
        callback(grantedStatus === PermissionsAndroid.RESULTS.GRANTED);
      } else {
        const result = await requestMultiple([
          PERMISSIONS.ANDROID.BLUETOOTH_SCAN,
          PERMISSIONS.ANDROID.BLUETOOTH_CONNECT,
          PERMISSIONS.ANDROID.ACCESS_FINE_LOCATION,
        ]);
        const isAllPermissionsGranted =
          result['android.permission.BLUETOOTH_SCAN'] ===
            PermissionsAndroid.RESULTS.GRANTED &&
          result['android.permission.BLUETOOTH_CONNECT'] ===
            PermissionsAndroid.RESULTS.GRANTED &&
          result['android.permission.ACCESS_FINE_LOCATION'] ===
            PermissionsAndroid.RESULTS.GRANTED;
        callback(isAllPermissionsGranted);
      }
    } else {
      callback(true);
    }
  };
`

VirtuozTM, 2023-01-30:
Hi @redko-o,

I finnaly found the issue and solved it. I was trying to ask the user to activate Bluetooth before I asked and got the BLUETOOTH_CONNECT permission.

So the solution is : 
```
const result = await requestMultiple([
            PERMISSIONS.ANDROID.BLUETOOTH_CONNECT,
            PERMISSIONS.ANDROID.BLUETOOTH_SCAN,
            PERMISSIONS.ANDROID.ACCESS_FINE_LOCATION,
          ]);
          if (
            result['android.permission.BLUETOOTH_CONNECT'] ===
            PermissionsAndroid.RESULTS.GRANTED
          ) {
            enableBluetooth();
```
wtih : 
`const enableBluetooth = async () => {
    const isBluetoothEnabled = await BluetoothStateManager.requestToEnable();
    if (!isBluetoothEnabled) {
      await BluetoothStateManager.enable();
    }
  };`

Btw, I use react-native-ble-plx and react-native-bluetooth-state-manager

I hope it will help you as much as possible

dominik-czupryna-withintent, 2023-09-29:
In Android 12+ the full BT management feature has been removed. Currently, it is no longer possible to turn BT on or off from the app. You can ask the user to turn it on, but this requires the BLUETOOTH_SCAN permission. The library supports this change since version 3.0.0.

closed 2023-09-29 by dominik-czupryna-withintent