dotintent/react-native-ble-plx #1050: Is this library compatible with React-Native "New Architecture"?

ablack34, 2023-01-25:
I may have to migrate a project to the latest version of React-Native with the "New Architecture". Is this library compatible with the New Architecture or is it planning on becoming compatible? 

Thanks!

fgrs, 2023-02-06:
I don't think this library works with RN 0.71.2 I get this error "cannot start scanning operation" and "could not find callback wrapper"

lucaswitch, 2023-02-06:
I guess it not supports since the last release was on 2021.

lucaswitch, 2023-02-06:
But it works on 0.71 apps completely fine.

ablack34, 2023-02-07:
Ok thanks, I'm assuming that means the maintainers aren't active? If that's the case is there a recommended maintained RN Bluetooth Library?

ScreamZ, 2023-02-07:
https://github.com/innoveit/react-native-ble-manager

closed 2023-02-08 by ablack34

bj97301, 2023-04-07:
> https://github.com/innoveit/react-native-ble-manager

Seems like it may require ejecting from expo.

gabimoncha, 2025-01-28:
@ablack34 it seems to be #1089