dotintent/react-native-ble-plx #1060: why the application slows down when scanning if it sees a lot of devices nearby (more than 100 pieces).

quadecco, 2023-03-10:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

How implement scan process? 
Every BLE device advertizes packet (for example manufactureData = ) every 1000 ms.
Is it problem in my side (organization code in JS)??
or
Scan process by BLE adapter takes some time and this big amount cause hanging??

quadecco, 2023-03-10:
forgot example manufactureData
4goUAQEREREAAItW1g8UUQ8AQlZ2UwAAAAA=

quadecco, 2023-03-10:
And main question
How to improve performance of my app??

otpz, 2023-07-19:
Have you been able to find a solution to this issue?

dominik-czupryna-withintent, 2023-09-29:
It depends on your use case but pay attention to two things:
- by default, duplicates are allowed, so you should check whether you are not detecting the same device multiple times
- if you are looking for a specific device in the first argument, you can pass the UUID of one or more of the target device services, and devices without those UUID will be filtered

closed 2024-02-07 by abursztynowska