dotintent/react-native-ble-plx #1064: Bug Report

smonikharaj, 2023-03-23:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed

## Expected Behavior

While Upgrading SDK version to 31, We have added the Bluetooth permission, While debugging locally it is working fine.
## Current Behavior

What is the current behavior?
Multiple times it is asking to turn on location even the location is on.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. At first we will start to accept the bluetooth
2. It will be asking to turn on the location
3. While using this function 'startDeviceScan' It says cannot start scanning operation.

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://dotintent.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 2.0.3
* Platform: Android/iOS.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

```js
Example Code: 
ScanForDevices(this.manager, errorMessage => {
      console.warn(errorMessage) ---- warning message is coming here only. saying 'Cannot start scanning operation'
      if (errorMessage) {
        if (Platform.OS === 'android') {
          AndroidOpenSettings.locationSourceSettings()
        } else {
          this.showLocationPermission()
        }
      }
    }, device => {
      if (device) {
        this.handleDiscoverDevices(device)
      }
    }, () => {
      setScanningState(DpsConstantTypes.STOP_DISCOVERING)
    })
    
    ********************************************************************************************
    In ScanforDevices function:
    
    try{
    manager.startDeviceScan([Service UUID], null, (error, device) => {
      if (error) {
        onError && onError(error.message)
      }
      if (device) {
        onSuccess && onSuccess(device)
      }
    })

    if (this.scanningTimeout !== null) {
      clearTimeout(this.scanningTimeout)
    }
    this.scanningTimeout = setTimeout(() => {
      try {
        manager.stopDeviceScan()
      } catch (error) {
        onError(error.message)
      }
      onScanComplete && onScanComplete()
    }, [timeout])
  } catch (error) {
    onError(error.message)
  }

```

encarvlucas, 2023-03-23:
I am on the team that caught this bug with Monikha. I need to make a small correction.

> What is the current behavior?
Multiple times it is asking to turn on location even the location is on.

This is due to the logic in the application, the real problem is that the bluetooth scan method `startDeviceScan` is not running succesfully and is returning the following error:
>Cannot start scanning operation

dominik-czupryna-withintent, 2023-09-29:
Could you also include the version of the system (iOS/Android) the device runs on?

closed 2023-11-02 by dominik-czupryna-withintent