dotintent/react-native-ble-plx #1067: Question: BleError: Characteristic xxx read failed for device xxx and service xxx

bj97301, 2023-04-07:
## Prerequisites

I am using `"react-native-ble-plx": "^2.0.3",`

## Question

Thank you in advance for your help 😁.  I really appreciate all the hard work you put into this amazing package ♥️.  

I am getting an error when running the below code after what seems to be a successful scan: 
```js

export let tryToConnectToDevice = async device => {
  console.log('try to connect to device')
  await manager().stopDeviceScan()
  events.triggerEvent(eventNames.searchingEnded)

  let d = await manager().connectToDevice(device.id)
  console.log('connected to device: ', bluetoothContracts.cleanUpBtShit(d))
  let dCheck = await manager().connectedDevices([serviceAndCharacteristicIds.service])
  console.log('connectedDevices: ', dCheck)

  let servicesAndCharacteristics = await d.discoverAllServicesAndCharacteristics()
  console.log(
    'servicesAndCharacteristics: ',
    bluetoothContracts.cleanUpBtShit(servicesAndCharacteristics),
  )
  let services = await manager().servicesForDevice(d.id)
  console.log('services: ', services)
  let characteristics = await manager().characteristicsForDevice(
    d.id,
    serviceAndCharacteristicIds.service,
  )
  console.log('characteristics: ', characteristics)

  if (characteristics[0]?.uuid == serviceAndCharacteristicIds.tx) {
    let characteristic = await manager().readCharacteristicForDevice(
      d.id,
      serviceAndCharacteristicIds.service,
      serviceAndCharacteristicIds.tx,
    )
    console.log('readCharacteristicForDevice: ', characteristic)
  } else {
    await manager().cancelDeviceConnection(d.id)
  }
}

```

console:
```bash

LOG  try to connect to device
 LOG  connected to device:  {"_manager": {"_activePromises": {}, "_activeSubscriptions": {"1": [Object]}, "_eventEmitter": {"_nativeModule": [Object]}, "_scanEventSubscription": null, "_uniqueId": 4}, "id": "xxx", "isConnectable": null, "localName": null, "manufacturerData": null, "mtu": 23, "name": "xxx", "overflowServiceUUIDs": null, "rssi": null, "serviceData": null, "serviceUUIDs": null, "solicitedServiceUUIDs": null, "txPowerLevel": null}
 LOG  connectedDevices:  [{"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 5}, "id": "xxx", "isConnectable": null, "localName": null, "manufacturerData": null, "mtu": 23, "name": "xxx", "overflowServiceUUIDs": null, "rssi": null, "serviceData": null, "serviceUUIDs": null, "solicitedServiceUUIDs": null, "txPowerLevel": null}]
 LOG  servicesAndCharacteristics:  {"_manager": {"_activePromises": {}, "_activeSubscriptions": {"1": [Object]}, "_eventEmitter": {"_nativeModule": [Object]}, "_scanEventSubscription": null, "_uniqueId": 7}, "id": "xxx", "isConnectable": null, "localName": null, "manufacturerData": null, "mtu": 23, "name": "xxx", "overflowServiceUUIDs": null, "rssi": null, "serviceData": null, "serviceUUIDs": null, "solicitedServiceUUIDs": null, "txPowerLevel": null}
 LOG  services:  [{"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 8}, "deviceID": "xxx", "id": 10791587712, "isPrimary": true, "uuid": "xxx"}]
 LOG  characteristics:  [{"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 9}, "deviceID": "xxx", "id": 10740079456, "isIndicatable": false, "isNotifiable": true, "isNotifying": false, "isReadable": false, "isWritableWithResponse": false, "isWritableWithoutResponse": false, "serviceID": 10791587712, "serviceUUID": "xxx", "uuid": "xxx", "value": "AK5tAw=="}, {"_manager": {"_activePromises": [Object], "_activeSubscriptions": [Object], "_eventEmitter": [NativeEventEmitter], "_scanEventSubscription": null, "_uniqueId": 9}, "deviceID": "xxx", "id": 10740079840, "isIndicatable": false, "isNotifiable": false, "isNotifying": false, "isReadable": false, "isWritableWithResponse": true, "isWritableWithoutResponse": true, "serviceID": 10791587712, "serviceUUID": "xxx", "uuid": "xxx", "value": null}]
 WARN  Possible Unhandled Promise Rejection (id: 0):
BleError: Characteristic xxx read failed for device xxx and service xxx
BleError: Characteristic xxx read failed for device xxx and service xxx
    at construct (native)
    at apply (native)
    at _construct (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:3348:28)
    at Wrapper (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:3310:25)
    at construct (native)
    at _createSuperInternal (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:103132:322)
    at call (native)
    at BleError (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:103140:26)
    at parseBleError (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:103170:30)
    at ?anon_0_ (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:103394:47)
    at throw (native)
    at asyncGeneratorStep (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:4644:26)
    at _throw (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:4666:29)
    at tryCallOne (/Users/distiller/react-native/sdks/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:53:16)
    at anonymous (/Users/distiller/react-native/sdks/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:139:27)
    at apply (native)
    at anonymous (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:29744:26)
    at _callTimer (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:29663:17)
    at _callReactNativeMicrotasksPass (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:29693:17)
    at callReactNativeMicrotasks (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:29856:44)
    at __callReactNativeMicrotasks (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:2793:46)
    at anonymous (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:2605:45)
    at __guard (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:2777:15)
    at flushedQueue (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:2604:21)
    at invokeCallbackAndReturnFlushedQueue (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=ios&dev=true&hot=false:2598:33)

```

bj97301, 2023-04-10:
Good question: `"react-native": "0.71.6",`

catolicasc, 2023-04-10:
I am having the same error, so I started using react-native-ble-manager, but I am not able to send messages to my Arduino =[
Do both libraries have some issue or is it just a configuration failure on my part?

bj97301, 2023-04-10:
I am actually in the process of doing the same. Was sad to have to eject for this.  I hope I am not wasting my time.

catolicasc, 2023-04-11:
I was able to use another library.. I suggest using it. `react-native-ble-manager,`

bj97301, 2023-04-11:
Oh awesome. Thank you.  I ran into different but similar issues with the second one. I’ll definitely try this one. 😃

bj97301, 2023-04-11:
Oh wait, that seems the same as the one you said you were having issues with as before. Did you figure it out?

bj97301, 2023-04-11:
I believe the plx should work with expo dev builds but the other one makes you eject. Both aren’t working for me though. Can’t get a value back from the characteristic calls I make. @catolicasc is there a truck you found out to make the non plx package read a characteristic’s value?

bj97301, 2023-04-11:
I am not sure. If I were you I’d use the non plx package as that one seems to be more up to date and supported. Especially if you already ejected from expo.

catolicasc, 2023-04-11:
```
import React, {useState, useEffect} from 'react';
import {
  View,
  Text,
  FlatList,
  NativeModules,
  NativeEventEmitter,
  TouchableOpacity,
  SafeAreaView,
  TextInput,
  ScrollView, PermissionsAndroid, Platform
} from "react-native";
import BleManager from 'react-native-ble-manager';
const BleManagerModule = NativeModules.BleManager;
const bleManagerEmitter = new NativeEventEmitter(BleManagerModule);
import {Buffer} from 'buffer';

export default function BluetoothScreen() {
  const buffer = Buffer.from('s');
  const [devices, setDevices] = useState([]);
  const [connectDevice, setConnectedDevice] = useState();
  const [text, setText] = useState();
  const [scanning, setScanning] = useState(false);
  const [infos, setInfos] = useState([]);
  const handleConnect = device => {
    setConnectedDevice(device);
    console.log(device);
    BleManager.connect(device.id).then(async () => {
      BleManager.retrieveServices(device.id)
        .then(async peripheralInfo => {
          const {characteristics} = peripheralInfo;
          setInfos({characteristic: characteristics});
        })
        .catch(error => {
          console.error(error);
        });
    });
  };
  useEffect(() => {
    console.log('scanear');
    BleManager.start({showAlert: false});

    const handleDiscoverPeripheral = device => {
      console.debug('dispositivo"', device);
      setDevices(prevDevices => {
        if (prevDevices.some(prevDevice => prevDevice.id === device.id)) {
          return prevDevices;
        }
        return [...prevDevices, device];
      });
    };

    BleManager.scan([], 7, true)
      .then(() => {
        console.log('Escaneando dispositivos');
      })
      .catch(error => console.error(error));

    bleManagerEmitter.addListener(
      'BleManagerDiscoverPeripheral',
      handleDiscoverPeripheral,
    );
    handleAndroidPermissions();

    return () => {
      // BleManager.stopScan();
      // bleManagerEmitter.removeListener('BleManagerDiscoverPeripheral', handleDiscoverPeripheral);
    };
  }, [scanning]);
  const handleAndroidPermissions = () => {
    if (Platform.OS === 'android' && Platform.Version >= 31) {
      PermissionsAndroid.requestMultiple([
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
      ]).then(result => {
        if (result) {
          console.debug(
            '[handleAndroidPermissions] User accepts runtime permissions android 12+',
          );
        } else {
          console.error(
            '[handleAndroidPermissions] User refuses runtime permissions android 12+',
          );
        }
      });
    } else if (Platform.OS === 'android' && Platform.Version >= 23) {
      PermissionsAndroid.check(
        PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
      ).then(checkResult => {
        if (checkResult) {
          console.debug(
            '[handleAndroidPermissions] runtime permission Android <12 already OK',
          );
        } else {
          PermissionsAndroid.request(
            PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
          ).then(requestResult => {
            if (requestResult) {
              console.debug(
                '[handleAndroidPermissions] User accepts runtime permission android <12',
              );
            } else {
              console.error(
                '[handleAndroidPermissions] User refuses runtime permission android <12',
              );
            }
          });
        }
      });
    }
  };
  const handleSend = async () => {
    console.log(connectDevice.id);

    console.log('infos', infos, connectDevice.id);
    BleManager.writeWithoutResponse(
      connectDevice.id,
      infos.characteristic[0].service,
      infos.characteristic[0].characteristic,
      buffer.toJSON().data
    )
      .then((data) => console.log('Mensagem enviada com sucesso!', buffer.toJSON().data))
      .catch(error => console.error('Erro ao enviar mensagem:',error));
  };

  return (
    <SafeAreaView style={{marginTop: 50, backgroundColor: '#fff'}}>
      {/*<TouchableOpacity*/}
      {/*  style={{padding: 10, backgroundColor: 'red'}}*/}
      {/*  onPress={() => {*/}
      {/*    setScanning(!scanning);*/}
      {/*  }}>*/}
      {/*  <Text>Scanear</Text>*/}
      {/*</TouchableOpacity>*/}
      <ScrollView style={{height: 200, borderWidth: 3}}>
        <FlatList
          data={devices}
          keyExtractor={device => device.id}
          renderItem={({item}) => (
            <TouchableOpacity
              onPress={() => handleConnect(item)}
              style={{padding: 5, backgroundColor: 'blue', marginBottom: 1}}>
              <Text style={{color: '#fff'}}>{item.id}</Text>
              <Text style={{color: '#fff'}}>{item.name}</Text>
              <Text style={{color: '#fff'}}>
                {' '}
                Serviços:{' '}
                {item?.advertising?.serviceUUIDs
                  ? item.advertising.serviceUUIDs.join(', ')
                  : 'vazio'}
              </Text>
              <Text style={{color: '#fff'}}>
                Características:{' '}
                {JSON.stringify(item.advertising.characteristicsUUIDs)}{' '}
                {item.advertising.characteristicsUUIDs
                  ? item.advertising.characteristicsUUIDs.join(', ')
                  : 'vazio'}{' '}
              </Text>
            </TouchableOpacity>
          )}
        />
      </ScrollView>
      <TextInput onChangeText={setText} />
      <TouchableOpacity onPress={handleSend} style={{marginBottom: 149}}>
        <Text>Enviar</Text>
      </TouchableOpacity>
    </SafeAreaView>
  );
}

```

catolicasc, 2023-04-11:
> @catolicasc : I tried your example but it doesn't work for me, could you post a video of how it works for you?

yes, a moment sir.

catolicasc, 2023-04-11:
https://drive.google.com/file/d/195EQiATdzcWfwH31M06XnT4xLO40DusY/view?usp=share_link

catolicasc, 2023-04-11:
refresh your useeffect

bj97301, 2023-04-12:
Oh your on android. Does it work on iPhone too?

catolicasc, 2023-04-12:
In my video I used in iphone 
( sorry my bad english )

catolicasc, 2023-04-12:
> In my video I used in iphone ( sorry my bad english )

Phisical iphone, Emulator not working.

bj97301, 2023-04-12:
@catolicasc your English is very good. Don’t worry 😃. Thank you for your help ❤️.

catolicasc, 2023-04-12:
> @bj97301: Não há problema para o seu inglês.
> 
> Mas se eu quiser enviar uma mensagem apenas para um dispositivo x, ele recebe?

yes

dominik-czupryna-withintent, 2023-09-29:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs. Please confirm if the issue still persists and create a new issue if it still does.

closed 2023-09-29 by dominik-czupryna-withintent

bj97301, 2023-09-29:
I believe I fixed it by doing things in the correct order. Been a while… shoulda closed this issue. Sorry bout that but Thank you.