dotintent/react-native-ble-plx #1068: Build Error : MultiplatformBleAdapter/MultiplatformBleAdapter.modulemap' not found

mehmettalhairmak, 2023-04-13:
## Question

I'm using Macbook Pro M2 (macOS 13.3) and Xcode 14.3. My React Native version is 0.66.4

I cloned my existing project to pc and when i try iOS building on Xcode, It returning `MultiplatformBleAdapter/MultiplatformBleAdapter.modulemap' not found` error.

I already opening project from .xcworkspace file. I deleted DerivedData folder and Clean Build Folder but this way couldnt solving my error

How can i successfully build my project?

I need a help.

My fully error message : `/Users/mehmettalhairmak/Library/Developer/Xcode/DerivedData/eighthsense-etwwmgaxtxfkgucpjsdynzflhdzq/Build/Intermediates.noindex/eighthsense.build/Debug-iphonesimulator/eighthsense.build/DerivedSources/eighthsense_vers.c module map file '/Users/mehmettalhairmak/Library/Developer/Xcode/DerivedData/eighthsense-etwwmgaxtxfkgucpjsdynzflhdzq/Build/Products/Debug-iphonesimulator/MultiplatformBleAdapter/MultiplatformBleAdapter.modulemap' not found
`

neelratanguria, 2023-06-23:
Hi @mehmettalhairmak ,

This is due to build fail and most probably due to pod dependency installation fail.

I found the best solutions is to have a remote Git repo. All the build files are generated on the local disk. So delete the local version of the repository and re-clone it. 

After that run `npm i` and then `npx pod-install`

It fixed my problem.

msakbar, 2023-09-17:
Hi,

Just chiming in here - i seem to have run into this issue and can't solve it either. Do you know if there's a build configuration item that I should be looking into?

dominik-czupryna-withintent, 2023-09-29:
I guessing you are asking about this error 
‘...path…/AwesomeProject_vers.c’ module map file '...path…/MultiplatformBleAdapter/MultiplatformBleAdapter.modulemap' not found
And I'm also guessing that you do the build via xCode.
To solve this error:
- install pods (in root project root `cd ios && pod install`
- open `.xcworkspace` instead of `.xcodeproj` (the white one not blue one)
[image]

closed 2023-09-29 by dominik-czupryna-withintent