dotintent/react-native-ble-plx #1069: How to implement the connection persistance with BLE device

AshishJ19, 2023-04-27:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

We need to implement connection persistance with BLE Device.

## Description
- First of all we are calling the `startDeviceScan` method to scan the device.
- After scanning successfully, we are calling `connectToDevice` method to initiate the connection with BLE Device.
- We are storing the `deviceId` in `AsyncStorage` after successful connection established.

## Use Case
We have to implement connection persistance with the previous connected BLE Device without scanning. To implement this we are calling `connectToDevice` and passing the previous connected BLE DeviceId. On some devices it is working fine but some devices it is not working fine because deviceId of ble device is being changed what we found. Any idea on this ?

moooss, 2023-05-12:
Hi,

Your approach on storing the deviceId is not the one to consider for several points :
 - New Ble specification  use dynamic MAC adresses
 - Android gives the MAC address for deviceId, but iOS gives an UUID
 - Scanning before each connection does not cost so much time

So I would advise to scan before connecting to the device.

Also, there is 2 subjects: 

1. Reconnect to a known device after closing the app. 
This is done by storing the ble name of the device, scan for it and reconnect.

2. Maintain connection in the app with screen navigation
For this, what I do is to check if the device is connecting in the lib everytime I want to use it. 
If it is the case you can get the device object from `manager.connectedDevices()`.
You might have to maintain the connection state (connecting, connected, disconnected) in the global state of your app.

Chears

dominik-czupryna-withintent, 2023-09-29:
I think what you want to do would require a device bound but we don't currently support this feature

closed 2023-09-29 by dominik-czupryna-withintent