dotintent/react-native-ble-plx #1070: Bug Report

bj97301, 2023-05-02:
## Expected Behavior
No warnings in the console:

## Current Behavior

```bash

 WARN  `new NativeEventEmitter()` was called with a non-null argument without the required `addListener` method. 
    at connectToController (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:133210:41)
    at withDevTools(connectToController) (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:102244:27)
    at RCTView
    at View (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:48591:43)
    at RCTView
    at View (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:48591:43)
    at AppContainer (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:48468:36)
    at main(RootComponent) (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:84982:28)
 WARN  `new NativeEventEmitter()` was called with a non-null argument without the required `removeListeners` method. 
    at connectToController (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:133210:41)
    at withDevTools(connectToController) (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:102244:27)
    at RCTView
    at View (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:48591:43)
    at RCTView
    at View (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:48591:43)
    at AppContainer (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:48468:36)
    at main(RootComponent) (http://192.168.1.3:8081/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.electroandcompany.trumoto&modulesOnly=false&runModule=true:84982:28)

````

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. open app on android
2. call manager constructor

### Context

I am getting this on "@config-plugins/react-native-ble-plx": "^6.0.0", and  "react-native": "0.71.7",

andyyou, 2023-05-26:
+1 the same problem but with RN 0.71.8 and not Expo project.

dominik-czupryna-withintent, 2023-09-29:
These errors should no longer appear (starting with 3.0.0).

closed 2023-09-29 by dominik-czupryna-withintent

bj97301, 2023-09-29:
Does this mean the maintainers are back?  Yus!!!  Let me know if I can help.