dotintent/react-native-ble-plx #1071: writeWithoutResponse does not resolve

pofffff, 2023-05-05:
- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

The writeCharacteristicWithoutResponseForDevice method should successfully send data to the connected Bluetooth device and resolve the Promise.

## Current Behavior

When attempting to use writeCharacteristicWithoutResponseForDevice, the code execution freezes, and no further code is executed. The Promise does not resolve or reject, even with the use of async/await, timeouts, and various other approaches.

### Steps to Reproduce

1. Connect to a Bluetooth headset using the react-native-ble-plx library.
2. Attempt to send data to the device using the writeCharacteristicWithoutResponseForDevice method.
3. Observe that the code execution freezes after the method is called and no further code is executed.

### Context

* Library version: 2.0.3
* Platform: Both iOS and Android, 
* JS logs: bleManager logging verbose
* Contents of the `package.json` file:
"dependencies": {
    "@react-native-async-storage/async-storage": "^1.17.11",
    "@react-native-community/netinfo": "^9.3.7",
    "@react-native-community/slider": "^4.3.3",
    "@react-navigation/native": "^6.0.13",
    "@react-navigation/native-stack": "^6.9.1",
    "axios": "^1.2.6",
    "base64-js": "^1.5.1",
    "react": "18.1.0",
    "react-native": "0.70.9",
    "react-native-ble-plx": "^2.0.3",
    "react-native-config": "^1.4.11",
    "react-native-device-check": "^1.0.0",
    "react-native-device-info": "^10.5.1",
    "react-native-document-picker": "^8.2.0",
    "react-native-fs": "^2.20.0",
    "react-native-gesture-handler": "^2.8.0",
    "react-native-linear-gradient": "^2.6.2",
    "react-native-safe-area-context": "^4.4.1",
    "react-native-screens": "^3.18.2",
    "react-native-splash-screen": "^3.3.0",
    "react-native-vector-icons": "^9.2.0",
    "react-native-wheel-scrollview-picker": "^2.0.1",
    "sinon": "^15.0.3",
    "spark-md5": "^3.0.2",
    "zustand": "^4.3.7"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@babel/runtime": "^7.12.5",
    "@react-native-community/eslint-config": "^2.0.0",
    "@tsconfig/react-native": "^2.0.2",
    "@types/jest": "^29.5.0",
    "@types/react": "^18.0.21",
    "@types/react-native": "^0.70.6",
    "@types/react-native-base64": "^0.2.0",
    "@types/react-native-vector-icons": "^6.4.12",
    "@types/react-test-renderer": "^18.0.0",
    "@types/sinon": "^10.0.14",
    "@types/spark-md5": "^3.0.2",
    "@typescript-eslint/eslint-plugin": "^5.37.0",
    "@typescript-eslint/parser": "^5.37.0",
    "babel-jest": "^29.5.0",
    "eslint": "^7.32.0",
    "jest": "^29.5.0",
    "metro-react-native-babel-preset": "^0.76.2",
    "react-native-version": "^4.0.0",
    "react-test-renderer": "18.1.0",
    "ts-jest": "^29.1.0",
    "typescript": "^4.8.3"
  }
  
* Formatted code sample or link to a repository:

```ts
function example(bytes: Uint8Array): Promise<boolean> {
  return new Promise(async (resolve, reject) => {
    console.log({bytes});
    if (dataCharacteristic) {
      console.log(dataCharacteristic);
      console.log('Sending RWCP segment');
      const base64 = convertUint8ArrayToBase64(bytes);
      console.log(base64);
      try {
        // @ts-ignore
        dataCharacteristic._manager
          .writeCharacteristicWithoutResponseForDevice(
            dataCharacteristic.deviceID,
            dataCharacteristic.serviceUUID,
            dataCharacteristic.uuid,
            base64,
            false,
          )
          .then(() => {
            console.log('Sent RWCP segment');
            resolve(true);
          })
          .catch(e => {
            console.error(e);
            reject(false);
          });
      } catch (e) {
        console.error(e);
        reject(false);
      }
    } else {
      reject(false);
    }
  });
}
```

justsee, 2023-05-27:
It's not shown in your code, but are you connecting to the device like this?

```
const bleManager = new BleManager()
const device = bleManager.connectToDevice(id)
await device.discoverAllServicesAndCharacteristics()
```

If you've connected to the device, and discovered the services and characteristics, then something like this should work:

```
const data = Buffer.from(bytes).toString('base64')
try {
    const res = await device.writeCharacteristicWithoutResponseForService(
      SERVICE,
      CHARACTERISTIC,
      data)
) catch (error) {
    console.log('error, JSON.stringify(error)
}
```

cordial, 2023-08-21:
@justsee doesn't work on ios ;)

dominik-czupryna-withintent, 2023-09-29:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs. Please confirm if the issue still persists. I tried to reproduce this problem on the latest package version but the function worked fine.

closed 2023-11-02 by dominik-czupryna-withintent

PrudhviBanki, 2024-01-23:
still issue in IOS