dotintent/react-native-ble-plx #1073: Getting "Scan failed because feature unsupported (code 8)" when calling startDeviceScan() for Android devices

jaewondong, 2023-06-05:
## Expected Behavior

Scan BLE peripherals with startDeviceScan()

## Current Behavior

Runs for IOS, but returns "Scan failed because feature unsupported (code 8)" for Android.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. ...

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://dotintent.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

- react: 17.0.2
- react-native: 0.67.3
- react-native-ble-plx: 2.0.3
- Platform: Android/iOS.
- Android Device: Pixel 2, Android 12 (API 31)

`bleManager.startDeviceScan(null, 
      {
        scanMode: ScanMode.LowLatency,
        callbackType: ScanCallbackType.FirstMatch,
      },
      async (error, device) => {
        if (error) {
          this.bleManager.stopDeviceScan();
          console.error('error scanning');
          console.log(JSON.stringify(error));
          rejectDevicePromise(error);
          return;
        }
        if (device?.localName === this.peripLocalName) { ... }`

thuantvDr4, 2023-06-28:
same issue, what's any solution? please

dominik-czupryna-withintent, 2023-09-29:
This is the error on the android side: https://developer.android.com/reference/android/bluetooth/le/ScanCallback#SCAN_FAILED_FEATURE_UNSUPPORTED
Most likely, the device has a problem with ScanMode.LowLatency. I tried to reproduce this problem on a phone with the same SoC but I didn’t find any problems. Could you confirm the issue after updating the package?

closed 2023-11-02 by dominik-czupryna-withintent