dotintent/react-native-ble-plx #1074: Is this library abandoned? No updates since Nov 2021

jamesrogers93, 2023-06-29:


TeddyGavi, 2023-07-07:
Hello! @jamesrogers93 I have recently spoken with the Marketing Manager from dotintent. He has told me that this library is no longer maintained but they have plans to re-open support in the near future. So at this point stay tuned!

bj97301, 2023-07-08:
Has anyone forked this and made some fixes?  I am in need of some reliability when connecting/disconnecting from devices =/

fabiendem, 2023-07-11:
`react-native-ble-plx` has been amazing, but the lack of maintenance is causing issues on Android 12 and above. (requires Location permission for a scan while that's not really needed anymore).
See as well https://github.com/dotintent/react-native-ble-plx/issues/908#issuecomment-1280663005

As mentioned by @TeddyGavi, https://github.com/innoveit/react-native-ble-manager has active support.
Its architecture is a bit simpler

@TeddyGavi 
> they have plans to re-open support in the near future

do you have any timeline please?

TeddyGavi, 2023-07-12:
@fabiendem That is interesting about the Android 12 and above issues. I never experienced those issues when working on my app. I used a different pattern than what is being done inside the `useEffect` on the thread in #908. 

> do you have any timeline please?

No, I wasn't given any more information by my contact at intent! Hopefully its soon!

pawlo1337, 2023-08-30:
Dear Friends, thank you for your patience. We will proceed crucial development works within next 2 months and the library will stay maintained.

closed 2023-08-30 by pawlo1337