dotintent/react-native-ble-plx #1077: Write failed: [BleError: Characteristic 00002a52-0000-1000-8000-00805f9b34fb write failed for device 00:5k:BF:C1:01:71 and service 00001808-0000-1000-8000-00805f9b34fb]

hembe001, 2023-08-14:
### Steps to Reproduce

let hexString = "0101"; // Assuming the initial value is not null
  const byteArray = hexString.match(/.{1,2}/g)?.map(byte => parseInt(byte, 16)) ?? [];
  const base64String = btoa(String.fromCharCode(...byteArray));
 manager.writeCharacteristicWithResponseForDevice(
                  device.id,
                   '00001808-0000-1000-8000-00805f9b34fb',
                   '00002A52-0000-1000-8000-00805f9b34fb', 
                   base64String,
                   false
                  )
                   .then((characteristic) => {
                   console.log('Write successful'+ JSON.stringify(characteristic));
                   })
                  .catch((error) => {
                     console.log('Write failed:', error);
                   });

1. I have used this code to call and send blood sugar readings successfully from a glucose monitoring device.
2. And then, while everything was going well, an error occurred.
3. Write failed: [BleError: Characteristic 00002a52-0000-1000-8000-00805f9b34fb write failed for device 00:5F:BF:C1:01:71 and service 00001808-0000-1000-8000-00805f9b34fb]

I would like to know what the issue is caused by and how to fix it.

dominik-czupryna-withintent, 2023-09-29:
Could you provide a larger code snippet showing how you connect to the device? What device has been used to test the connection (iOS/Android)? Have you excluded a problem with the device itself by connecting to it via a BLE testing application like nRF connect?

closed 2023-11-02 by dominik-czupryna-withintent

Yangnnnn, 2024-03-28:
same issue